ukBaz/python-bluezero #16: add bluez build and install instructions to README or INSTALL

WayneKeenan, 2016-04-09:
One day this hopefully wont't be necessary, but for now it would be good to have a recipe on how to setup bluez

So for example, this is what I have in my notes but needs tweaking (specifically, i'm not convinced I've got all the paths in the configure step covered) so I've added this as a documentation request rather than a pull.

### Install build and install dependancies:
```
sudo apt-get install libglib2.0-dev libdbus-1-dev libical-dev libusb-1.0-0.dev libreadline-dev checkinstall
```

### Configure source 
```
./configure \
        --libdir=/lib \
        --libexec=/lib \
        --enable-experimental \
        --disable-cups 
```

### Perform an install 
This is performed as a removable debian package as opposed  to `make install` which does a blind file copy that is hard to remove.

```
sudo systemctl stop bluetooth
sudo checkinstall
```

### Enable the experimental option when the daemon starts
```
sudo sed -i '/^ExecStart.*bluetoothd\s*$/ s/$/ --experimental/' /lib/systemd/system/bluetooth.service
```

### rLight the blue touch paper
```
sudo systemctl daemon-reload && sudo systemctl restart bluetooth && sudo systemctl status bluetooth
sudo hciconfig hci0 up
hci config
```

WayneKeenan, 2016-04-09:
Perhaps of note in all the above is that the generated debian file satisfies the Raspian Jessie 'bluez' dependancy of the 'pi-bluetooth' package (amongst others).  But as mentioned the *.deb  /shouldn't really be passed around.  It's perhaps a very good start point of creating a proper debian package.

ukBaz, 2016-04-10:
Initial document created and added to repository

closed 2016-04-10 by ukBaz