ukBaz/python-bluezero #20: send_notify_event() only supports a single byte

WayneKeenan, 2016-04-30:
It should support up to at least the standard 20 bytes payload.

WayneKeenan, 2016-04-30:
the fix might be as simple as:

```diff
diff --git a/bluezero/peripheral.py b/bluezero/peripheral.py
index 4ae3cc0..8054f2c 100644
--- a/bluezero/peripheral.py
+++ b/bluezero/peripheral.py
@@ -330,7 +330,7 @@ class Characteristic(dbus.service.Object):
         # print('Update prop')
         self.PropertiesChanged(
             GATT_CHRC_IFACE,
-            {'Value': [dbus.Byte(self.value)]}, [])
+            {'Value': self.value}, [])
 ```

ukBaz, 2016-04-30:
Thanks for raising this.
I suspect that it needs to be changed to a `dbus.ByteArray`. I will take a look

ukBaz, 2017-08-12:
@WayneKeenan can we close this? is this still a problem?
If it is, can you create a simple test?

closed 2017-11-28 by ukBaz