ukBaz/python-bluezero #30: Bluetooth LE on a Pi3 running Raspian - How to get it working

WayneKeenan, 2016-05-26:
Prior to the May'16 Raspbian release with the 4.4 Kernel (which needs the steps below applied) my last 'known good' versions were:

https://downloads.raspberrypi.org/raspbian/images/raspbian-2016-02-29/2016-02-26-raspbian-jessie.zip
sudo rpi-update 692dde0c1586f7310301379a502b9680d0c104fd
BlueZ: 5.39

To get the bluezero `light_switch` example working on the latest Raspian I had to make a few tweaks involving:

 - Installing Bluez 5.39
 - Patching the BlueZ `hciattach`
 - config.txt changes
 - Running the bluezero examples with `sudo` (which I didn't thing was necessary before)

## Patch `hciattach`

Reference: see: https://github.com/libretro/Lakka/issues/314

```diff
diff -Naur a/tools/hciattach_bcm43xx.c b/tools/hciattach_bcm43xx.c
--- a/tools/hciattach_bcm43xx.c	2016-02-25 18:23:15.468717720 +0000
+++ b/tools/hciattach_bcm43xx.c	2016-02-25 18:24:08.028997243 +0000
@@ -368,9 +368,6 @@
 	if (bcm43xx_locate_patch(FIRMWARE_DIR, chip_name, fw_path)) {
 		fprintf(stderr, "Patch not found, continue anyway\n");
 	} else {
-		if (bcm43xx_set_speed(fd, ti, speed))
-			return -1;
-
 		if (bcm43xx_load_firmware(fd, fw_path))
 			return -1;
 
@@ -379,6 +376,7 @@
 			perror("Can't set host baud rate");
 			return -1;
 		}
+		sleep(1);
 
 		if (bcm43xx_reset(fd))
 			return -1;
```

## Firmware files

Starting the patched `hciattach` gives an error about not being able to find `/etc/firmware`

```
sudo ln -sf /lib/firmware/ /etc/firmware
```

There's probably a way to configure this at BlueZ `./configure` time.

## config.txt

This is my entire `/boot/config.txt` which includes some commented out lines tried whilst investigating, things worked better without them in the end.

```
dtparam=audio=on
lcd_rotate=2
start_x=1
gpu_mem=128
framebuffer_width=1280
framebuffer_height=720
#core_freq=250
#enable_uart=1
#dtoverlay=pi3-miniuart-bt
```

WayneKeenan, 2016-05-26:
I've just found some additional BlueZ patches by of the Pi maintainers.

```
cd bluez-5.39
wget https://gist.github.com/pelwell/c8230c48ea24698527cd/archive/3b07a1eb296862da889609a84f8e10b299b7442d.zip
unzip 3b07a1eb296862da889609a84f8e10b299b7442d.zip
git apply  -v c8230c48ea24698527cd-3b07a1eb296862da889609a84f8e10b299b7442d/*
```

Then configure, compile and install BlueZ as per the `bluezero` instructions.

WayneKeenan, 2016-05-26:
Using `sudo` on the latest (May'16) Raspbian is currently required.

I've found a there is a DBUS ACL difference between the Feb'16 and May'1 releases, the patch is:

To add the ACL setting back:

```diff
--- bluetooth.conf	2016-05-26 06:58:37.013356724 +0000
+++ bluetooth_addgroup.conf	2016-05-26 06:58:59.884183163 +0000
@@ -24,6 +24,11 @@
     <allow send_interface="org.freedesktop.DBus.Properties"/>
   </policy>
 
+  <!-- allow users of bluetooth group to communicate -->
+  <policy group="bluetooth">
+    <allow send_destination="org.bluez"/>
+  </policy>
+
   <policy at_console="true">
     <allow send_destination="org.bluez"/>
   </policy>
```

Apply the patch

```
sudo patch -b /etc/dbus-1/system.d/bluetooth.conf < dbus-bluetoothgroup.patch
```

Add the current user (e.g. pi) to the `bluetooth` group  (re-login required)

```
sudo usermod -G bluetooth -a $USER
```

ukBaz, 2016-09-29:
This appears to have been fixed with the combination of:
Raspbian Jessie with PIXEL 
Version: September 2016
Release date: 2016-09-23

And was tested along with BlueZ 5.42

ukBaz, 2016-09-29:
After a reboot the problem seems to be back so I'm re-opening the issue
```
$ bluetoothctl 
[NEW] Controller AA:AA:AA:AA:AA:AA RPi3 [default]
[bluetooth]# show
Controller AA:AA:AA:AA:AA:AA
	Name: RPi3
	Alias: RPi3
	Class: 0x000000
	Powered: yes
	Discoverable: no
	Pairable: yes
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d052A
	Discovering: no
[bluetooth]# exit
[DEL] Controller AA:AA:AA:AA:AA:AA RPi3 [default]
```

The is some noise up above as to exactly what steps are needed so I summarising the steps I took:
```
$ cd bluez-5.42/
$ wget https://gist.github.com/pelwell/c8230c48ea24698527cd/archive/3b07a1eb296862da889609a84f8e10b299b7442d.zip
$ unzip 3b07a1eb296862da889609a84f8e10b299b7442d.zip
$ git apply  -v c8230c48ea24698527cd-3b07a1eb296862da889609a84f8e10b299b7442d/*
$ make && sudo make install
$ sudo reboot
```

bryanmcgrane, 2016-12-30:
Anyone know if this issue has been resolved? I'm going to try the steps above with my Pi3 and report back.

ukBaz, 2016-12-30:
@bryanmcgrane I believe that there is no plan to send this patch upstream to BlueZ. This means that if you update BlueZ on a RPi3 then you will need to make sure you apply the patch. I believe that @pelwell knows that his gist is being used and has no plans to remove it.

bryanmcgrane, 2016-12-30:
I was actually able to get everything running with a few modified steps. I submitted the question on Stack Overflow and answered my own question. I was able to use Bluez v5.43.

http://stackoverflow.com/questions/41351514/leadvertisingmanager1-missing-from-dbus-objectmanager-getmanagedobjects-solved/41398903#41398903

ukBaz, 2016-12-30:
@bryanmcgrane your instructions on Stack Overflow seem very similar to ones at:
https://github.com/ukBaz/python-bluezero/blob/master/docs/install_bluez.rst

It would be interesting to know if the `bcm43xx-3wire` entry is in `tools/hciattach.c` without the applying the patch from pelwell. More details about the patch are at:
https://gist.github.com/pelwell/c8230c48ea24698527cd

I'd also be interested to see what the address of the Controller is reported as when your run `bluetoothctl` with your setup.

bryanmcgrane, 2016-12-30:
Yes my process is very similar, but I ended up having to omit the configuration and make options.

bluetoothctl can't find a default adapter for me. I can check my BT address using hciconfig however which shows

`hci0:   Type: Primary  Bus: UART
        BD Address: B8:27:EB:88:0E:F3  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
        RX bytes:4603 acl:141 sco:0 events:186 errors:0
        TX bytes:5064 acl:128 sco:0 commands:102 errors:0
`

pelwell, 2016-12-30:
Although we added the 3wire option, we aren't using it because we found the bcm43xx H5 support to be unreliable.

ukBaz, 2016-12-30:
Thanks for the reply @pelwell .

However this has confused me because using an updated BlueZ version doesn't work without the patch. Do you understand why this might be?

pelwell, 2016-12-30:
Another function of the patches is to upload the firmware at a low baud rate to make it reliable in the absence of flow control - I think that is why you need  at least one of them.

ukBaz, 2017-08-16:
Just adding a note here that there is an active discussion going on about the AA:AA:.. address issue over on the BlueZ developer mailing list:
http://marc.info/?l=linux-bluetooth&m=150289514416084&w=2

ukBaz, 2018-10-04:
Closing this as I think the issue has been resolved with newer versions of BlueZ

closed 2018-10-04 by ukBaz