ukBaz/python-bluezero #31: ERROR:dbus.connection:Unable to set arguments

WayneKeenan, 2016-05-26:
If you see an error such as the one mentioned in the title of this issue and perhaps looking a bit similar to this:

```
ERROR:dbus.connection:Unable to set arguments (dbus.ObjectPath('/ukBaz/bluezero/advertisement1978969712'), {}) according to signature None: <type 'exceptions.ValueError'>: Unable to guess signature from an empty dict
```

It's probably due to an incompatibility between the (likely 'old') version of BlueZ installed and running on your system and the expectations of the DBus client Python code in `bluezero` .

You should first check that the version your installed BlueZ is 'recent' and/or supported by `bluezero`, current 5.39

To check your current version run:
```
bluetoothd --version
```
or
```
/usr/sbin/bluetoothd
```

closed 2016-05-26 by WayneKeenan

LJBD, 2017-07-19:
I've run into this problem with Bluez 5.41 (native to CentOS 7 that I'm using). Can anything else be done here or do I have to downgrade Bluez?

ukBaz, 2017-07-19:
It is difficult to give specific answers based on the information you've given.

If I was to speculate, I would suspect "downgrading" is the wrong direction. Newer is generally the better direction with Bluez

I think the GATT server functionality was still behind the experimental switch in 5.41. Have you looked at: https://github.com/ukBaz/python-bluezero/blob/master/docs/install_bluez.rst#automatically-run-bluetoothd-with-experimental-mode
?

LJBD, 2017-07-23:
@ukBaz - Thanks for the suggestion & sorry for not giving you more information in the first place.
I will try to elaborate a bit more now.

I had to develop a GATT service on CentOS 7 but I'm quite new to the whole BLE thing, so I looked for a Python library that would get the job done for me. That's also why I've asked that question in #126, since I wanted to get your opinion on the matter (which was exhaustive, thanks for that as well!).

The problem was that I only have a Windows 10 laptop with virtual Linux machines.
At first, I've tried connecting to Bluetooth with a VM using VMWare Player, but it turned out, that it only supports Bluetooth 2.1 - I've checked that with `hciconfig -a`.

That's also where this `ERROR:dbus.connection` happened - I'm guessing that it couldn't create a GATT service, because the Bluetooth controller didn't support it.

Then, I've found information that you can connect the hardware bluetooth controller from a Windows host to a Linux guest VM if you use VirtualBox. There's [a good how to](https://forums.virtualbox.org/viewtopic.php?f=3&t=83325) on that on the VirtualBox forum.

In the end, I was able to successfully use your `cpu_temperature.py` example (after I've removed all code for the actual reading of temperature and left only the random numbers) and then adapt it to my own needs.

I have one more question regarding that: I want to send a float value with % as a unit and my client doesn't seem to understand that. I use the following format:
```python
format_value = dbus.Array([dbus.Byte(0x14),  # format: 32 bit float
                               dbus.Byte(0x00),  # exponent
                               dbus.Byte(0xAD),  # LSB of unit (%)
                               dbus.Byte(0x27),  # MSB of unit (%)
                               dbus.Byte(0x01),  # namespace
                               dbus.Byte(0x00),  # LSB of description
                               dbus.Byte(0x00)])  # MSB of description
```
Can you please look at that and tell me whether it's a good format or not?

I also have a suggestion regarding packing `sint16` in the `cpu_temperature.py` example - you can use `struct.pack` to do that for you. Here's how I pack the float value that I have:
```python
def float_to_bytes(value):
    """
    Converts a float to bytes according to IEEE standard (there's a nice
    description at `Wikipedia
    <https://en.wikipedia.org/wiki/Single-precision_floating-point_format>`_).
    :param value: a float value to be turned into bytes
    :return: a byte array with the converted value
    """
    ba = bytearray(struct.pack("<f", value))
    return ba

def to_bytes(value):
    answer = []
    try:
        value_float32 = float_to_bytes(value[0])
    except TypeError:
        value_float32 = float_to_bytes(value)
    for byte in value_float32:
        answer.append(dbus.Byte(byte))
    return answer
```

ukBaz, 2017-07-23:
Thanks @LJBD for sharing the details of how you got it working. I really appreciate it when people post the details of how they got things working.

Right, on to your additional question:

An IEEE-754 32-bit floating point is one of the formats talked about in the Bluetooth specification and summarized at:
https://www.bluetooth.com/specifications/assigned-numbers/format-types
You would use the `0x2904` descriptor with a value of `0x14` to tell the client you are sending an IEEE-754 32-bit floating point. You seem to have worked this out already as you have updated the `format_value` to this.
The next question is, does your client look for (and use) the Characteristic Presentation Format?

The other thing to notice is that I've given the CPU temperature characteristic the assigned number of `0x2A6E` which is a [Temperature characteristic ](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.temperature.xml) with an exponent of -2.
You should change that assigned number if you are not going to be sending a temperature with an exponent of -2.
I cannot see a [characteristic number ](https://www.bluetooth.com/specifications/gatt/characteristics) already assigned with a suitable definition for your use case so you will need to use a custom assigned number.
There is a range you have to avoid with your custom 128-bit UUID but I cannot find that this morning. If you need more information on this then open a new issue and I'll look harder :-)

Finally,
```python
ba = bytearray(struct.pack("<f", value))
```
only creates a float32 so I would use that in the function name.

LJBD, 2017-07-23:
Thanks for the answer, @ukBaz!

> does your client look for (and use) the Characteristic Presentation Format?

Well, I don't know for sure, but it was able to see that the format of `cpu_temperature` was `sin16` with Celsius degrees as the uint. I'm using [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) for Android as a client.

However, it might have understood the temperature characteristic number `0x2A6E` and that's why it processed the presentation format correctly.

> You should change that assigned number if you are not going to be sending a temperature with an exponent of -2.

Yes, I've changed that to a custom one. It was given to me by someone a lot more knowledgeable on BLE so I suppose it's outside of the banned range :)

> I would use that in the function name

Of course, that makes sense 👍