ukBaz/python-bluezero #39: Support Characteristic Presentation Format descriptor (0x2904)

ukBaz, 2016-06-06:
This descriptor is composed of five parts: format, exponent, unit, name space and description. The Format field determines how a single value contained in the Characteristic Value is formatted. The types of format supported are:

Format | Short Name | Description | Exponent Value
-------|------------|-------------|---------------
0x00 | rfu | Reserved for future use | No
0x01 | boolean | unsigned 1 - bit; 0 = false, 1 = true | No
0x02 | 2bit | unsigned 2 - bit integer | No
0x03 | nibble | unsigned 4 - bit integer | No
0x04 | uint8 | unsigned 8 - bit integer | Yes
0x05 | uint12 | unsigned 12 - bit integer | Yes
0x06 | uint16 | unsigned 16 - bit integer | Yes
0x07 | uint24 | unsigned 24 - bit integer | Yes
0x08 | uint32 | unsigned 32 - bit integer | Yes
0x09 | uint48 | unsigned 48 - bit integer | Yes
0x0A | uint64 | unsigned 64 - bit integer | Yes
0x0B | uint128 | unsigned 128 - bit integer | Yes
0x0C | sint8 | signed 8 - bit integer | Yes
0x0D | sint12 | signed 12 - bit integer | Yes
0x0E | sint16 | signed 16 - bit integer | Yes
0x0F | sint24 | signed 24 - bit integer | Yes
0x10 | sint32 | signed 32 - bit integer | Yes
0x11 | sint48 | signed 48 - bit integer | Yes
0x12 | sint64 | signed 64 - bit integer | Yes
0x13 | sint128 | signed 128 - bit integer | Yes
0x14 | float32 | IEEE - 754 32 - bit floating point | No
0x15 | float64 | IEEE - 754 64 - bit floating point No
0x16 | SFLOAT | IEEE - 11073 16 - bit SFLOAT | No
0x17 | FLOAT | IEEE - 11073 32 - bit FLOAT | No
0x18 | duint16 | IEEE - 20601 format | No
0x19 | utf8s | UTF - 8 string | No
0x1A | utf16s | UTF - 16 string | No
0x1B | struct | Opaque structure | No

This means that numbers need to be presented correctly to the dbus ReadValue and also converted from WriteValue. My proposal is to have a series of utility functions like the following:
```
def int_to_dbus_uint16(value):
    as_bytes = value.to_bytes(
        (value.bit_length() // 8) + 1, byteorder='little')
    return_val = []
    for byte in as_bytes:
        return_val.append(dbus.Byte(byte))
    return return_val
```

WayneKeenan, 2016-06-08:
Bluezero could be ported to other platforms (e.g. Mac) that do not use D-Bus to access the native BLE interface. 
Does it  make sense for `bluezero` not to use the text 'dbus' in an of its API and in doing so hide any 'dbus' API calls behind very thin veneers/facades functions to make bluezero platform agnostic?

So, for example, instead of:
```
bus.Byte(byte)
```
use;
```
bluezero.Byte(byte)
```

ukBaz, 2016-06-08:
Agreed that I should look to minimise the use of dbus.Byte() and definitely shouldn't put dbus in the function name. 
However this library is by design very closely linked to Bluez's dbus API so I'm not sure if it will be easy to separate the two.

ukBaz, 2016-10-21:
Closed by d1be4f9b015eace6a196b12b63e6878f4d921bf1

closed 2016-10-21 by ukBaz