ukBaz/python-bluezero #79: Unable to run fatbeacon example

nimble99, 2016-10-10:
I have downloaded bluezero from github (main), and also installed bluez 5.42 from source.

When I run sudo python fatbeacon.py, I get:

```
ERROR:dbus.connection:Unable to set arguments (u'/ukBaz/bluezero/advertisement0000', {}) according to signature None: <type 'exceptions.ValueError'>: Unable to guess signature from an empty dict
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 604, in msg_reply_handler
    reply_handler(*message.get_args_list(**get_args_opts))
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 401, in _introspect_reply_handler
    self._introspect_execute_queue()
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 387, in _introspect_execute_queue
    proxy_method(*args, **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 137, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 584, in call_async
    message.append(signature=signature, *args)
ValueError: Unable to guess signature from an empty dict
GATT application registered
```

What could be the cause?

Thanks

ukBaz, 2016-10-10:
Hi [Adam](@nimble99),

I can get the same error message as this if I run without the `--experimental` flag on the bluetooth daemon. This error is because the Advertisement DBus API is still considered beta by BlueZ.

The instructions for switching on experimental mode are available at:
http://pythonhosted.org/bluezero/install_bluez.html#automatically-run-bluetoothd-with-experimental-mode
If you run `service bluetooth status` you should see the experimental flag e.g:
```
$ service bluetooth status 
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled)
   Active: active (running) since Mon 2016-10-10 06:01:02 UTC; 5s ago
     Docs: man:bluetoothd(8)
 Main PID: 1441 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─1441 /usr/libexec/bluetooth/bluetoothd --experimental
```
Also, there shouldn't be a requirement to run with sudo. Here is how I'm running this:
```
$ python examples/level100/fatbeacon.py 
Advertisement registered
GATT application registered
```
If you are required to run with sudo you might have run `make` with sudo. You should only need to run `make install` with sudo.

Let me know how you get on...

Regards,
Barry
P.S. I normally run with Python3 although this should work with either.

ukBaz, 2016-10-21:
Hi [Adam](@nimble99),

I've not heard an update from you so I'm going to assume it is working now.
If that is not the case then please re-open the ticket.

Thanks,
Barry

closed 2016-10-21 by ukBaz