ukBaz/python-bluezero #84: Unable to run level10/light_switch.py

WayneKeenan, 2016-10-11:
bluez 5.42
raspberrypi
python3 
bluezero/examples/level10/light_switch.py

```
Traceback (most recent call last):
  File "/home/pi/pycharm.projects/bluezero/examples/level10/light_switch.py", line 96, in <module>
    app = peripheral.Application()
  File "/home/pi/pycharm.projects/bluezero/bluezero/peripheral.py", line 132, in __init__
    self.dongle = adapter.Adapter(device_id)
  File "/home/pi/pycharm.projects/bluezero/bluezero/adapter.py", line 88, in __init__
    self.path)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 244, in __init__
    _dbus_bindings.validate_object_path(object_path)
TypeError: must be str, not None

```

WayneKeenan, 2016-10-11:
I made a few tweaks to the example and peripheral.py to get it working:

```diff
diff --git a/bluezero/peripheral.py b/bluezero/peripheral.py
index ec5e0c0..c39a41c 100755
--- a/bluezero/peripheral.py
+++ b/bluezero/peripheral.py
@@ -128,7 +128,7 @@ class Application(dbus.service.Object):
 
         # Initialise services within the application
         self.services = []
-
+        device_id = device_id or adapter.list_adapters()[0]
         self.dongle = adapter.Adapter(device_id)
 
     def get_path(self):
@@ -208,7 +208,7 @@ class Application(dbus.service.Object):
         ``try-except-finally`` block to enable keyboard interrupts.
         """
         # Register the Bluetooth adapter
-        self.dongle.powered('on')
+        self.dongle.powered = True
 
         # Setup the advertising manager
         print('setup ad_manager')
diff --git a/examples/level10/light_switch.py b/examples/level10/light_switch.py
index 67c6cb3..92db92e 100644
--- a/examples/level10/light_switch.py
+++ b/examples/level10/light_switch.py
@@ -96,7 +96,7 @@ for mngd_objcs in light_service.GetManagedObjects():
 app = peripheral.Application()
 app.add_service(light_service)
 
-app.add_device_name('BluezeroLight')
+#app.add_device_name('BluezeroLight')
```

ukBaz, 2016-10-21:
fixed by d1be4f9b015eace6a196b12b63e6878f4d921bf1

closed 2016-10-21 by ukBaz