ukBaz/python-bluezero #90: Consider renaming this project

WayneKeenan, 2016-10-25:
If you google bluezero you are presented with images NSFW, let alone a younger audience.

WayneKeenan, 2016-10-25:
bluetoothzero ?

ukBaz, 2016-10-25:
That is a very undesirable situation. I did check before using the name and the pictures did not turn up then. They still do not show on other search engines like duckduckgo.com although I have to accept that Google is the market leading search engine. They also do not show up if I use bluezero and another keyword (which is maybe why this has not arisen before). 
To start this project I had to go through a rather lengthy legal process to ensure this was clearly an open source project under an MIT license. I have reread the agreement this evening and I believe a rename would require me to go through that legal process first and as a result I can not change anything in the short term. This feels like I'm between a rock and a hard place especially as the 'official' name is Python-Bluezero which gives reassuringly family friendly results.
I welcome opinions from you and anyone else following the project although for now (after nine hours of agonising over it) I'm going to close this as I can not justify the time and cost to change the name.

closed 2016-10-25 by ukBaz

WayneKeenan, 2016-10-26:
I've noticed that on a Desktop browser there is a 'report image' button on the top right of the Google results.
I've been reporting all the NSFW images, multiple times (not sure this will work in the same browser), but please join in and demote the results!