ukBaz/python-bluezero #94: Notifications not working

schef, 2016-11-07:
Or i am not sure how to use them.

ukBaz, 2016-11-29:
Apologies @schef I didn't see this issue get raised.

Are you asking about this in the peripheral or central role?

The [cpu_temperature](https://github.com/ukBaz/python-bluezero/blob/master/examples/level100/cpu_temperature.py) is an example of the peripheral role. 

For the central role I don't seem to have an example although in the [GATT](https://github.com/ukBaz/python-bluezero/blob/master/bluezero/GATT.py#L197) file you can see that you need to connect to the `PropertiesChanged` signal on the DBus.

Let me know some more information about what you are trying to do and I'll give you some more help.

ukBaz, 2017-09-03:
There has been no update on this so closing. Please re-open if it is still an issue

closed 2017-09-03 by ukBaz

mamert, 2019-06-14:
Hi, Barry, what do you mean by "connect to the PropertiesChanged signal on the DBus"?
As far as I can tell add_characteristic_cb does that.

Still, I try to connect to an ESP32 running this: https://github.com/mamert/iot-b/blob/master/fake_bt_device/fake_bt_device.ino

... with the following code:
```
import time
from bluezero import central
import logging

def on_notify(self, iface, changed_props, notused):
    print('on_notify')

#start here
logger = logging.getLogger()
logger.addHandler(logging.StreamHandler())

scale = central.Central(device_addr='C8:B2:1E:40:E9:8B')

characteristic = scale.add_characteristic(srv_uuid='0000fff0-0000-1000-8000-00805f9b34fb',
                        chrc_uuid='0000fff1-0000-1000-8000-00805f9b34fb')

scale.connect()
characteristic.add_characteristic_cb(on_notify)
characteristic.start_notify()

while scale.connected:
    val = characteristic.value
    print('Read')
    time.sleep(1.1)
```

... and only reads work, no notifications (no errors, but he callback isn't called).
I don't have a micro:bit to compare with.
Am I missing something?

ukBaz, 2019-06-14:
On the move at the moment and doing this on my phone so if the following doesn't help, the open a new issue and I'll provide a longer answer.

For notifications you need an event loop. In Bluezero there are some examples called async. Not quite what you are looking for which is why there is #190

mamert, 2019-06-14:
So, all I was missing was `scale.run()`
Thank you :)