ukBaz/python-bluezero #96: Can't write characteristic value from iPhone

wuyj88, 2016-11-11:
HI, thank you very much for the sharing.

I am building a application between RPi 3 and iPhone(iPhone is central), all the things are good, but when I try to write the value of characteristic using function:
[self.selectPeripheral writeValue:data forCharacteristic:self.selectCharacteristic type:CBCharacteristicWriteWithResponse];

It can not go through and the ERROR description is below:
ERROR:dbus.service:Unable to append (None,) to message with signature v: <type 'exceptions.TypeError'>: Don't know which D-Bus type to use to encode type "NoneType"

Several days passed, still didn't figure out.

Any help will be appreciated!

ukBaz, 2016-11-11:
Are you initializing the value of that characteristic before you write the value? 

For example, if you look at the light switch example I test for `None`. If it is `None` then write a `0` value in the callback as I was seeing a similar error to what you are getting if I didn't
https://github.com/ukBaz/python-bluezero/blob/master/examples/level10/light_switch.py#L23

This is because dbus was expecting to see an integer value and `None` is not an integer.

wuyj88, 2016-11-11:
Thanks for the reply.

I build my application based on the light switch example under folder level100 of version 0.0.2.
I initial the characteristic as following:
self.light = localGATT.Characteristic(1,
                                                LIGHT_UUID,
                                               self.srv,
                                                [0],
                                               True,
                                               ['read','write', 'notify'])
self.switch = localGATT.Characteristic(2,
                                                 SWITCH_UUID,
                                                 self.srv,
                                                 [0],
                                                 True,
                                                 ['read', 'write', 'notify'])
 Then use my iOS app to read and write the value, read is ok, but write is not. I am not quite understand about dbus expecting to see an integer because when I use characteristic.WriteValue('hello', '') on RPi side, it is also fine, because when I read it again, the string can be printed out.

I tried lightswitch you mentioned above using blue zero 0.0.1, it even can not be read by my iOS app and LightBlue, confused..

ukBaz, 2016-11-12:
OK. I'm at the airport currently so I can't do any testing right now. Couple of things to try while you are waiting...
1) are you sending a byte array from lightblue?
2) try to monitor what value is being sent over DBus.
With this
https://github.com/ukBaz/python-bluezero/blob/master/docs/install_bluez.rst#monitor-dbus-traffic
Or this
https://gist.github.com/ukBaz/d7cd0c4b9e7078c89980a3db2bbad98b

ukBaz, 2016-11-13:
@wuyj88 can you confirm which version of BlueZ you are using?
A simple way to do this from the command line is:
`bluetoothctl --version`

I can reproduce your error message if I run on BlueZ 5.40.
The current master is meant to run with BlueZ 5.42 or later. 

Having said that I'm seeing a different issue with running on the later version. I'll continue to investigate

wuyj88, 2016-11-13:
Hi, I am using bluez 5.42, I will give you a description in detail later since it is not convenient now.

Thanks

wuyj88, 2016-11-14:
It seems RPi received the value but the call back function can not be invoked due to the parameter with signature 'v', for example, I write 'A' and 'F', the error message is as below, you can see dbus.Array([dbus.Byte(10)], and dbus.Array([dbus.Byte(15)] have the right decimal value included.

Char Prop initialized
Char Prop initialized
Char Prop initialized
Char Prop initialized
Char Prop initialized
Char Prop initialized
Char Prop initialized
Char Prop initialized
Char Prop initialized
Char Prop initialized
Device added at /org/bluez/hci0/dev_49_C3_C7_AC_75_F4
GATT application registered
Advertisement registered
Property changed [dbus.String(u'Connected')]
Property changed [dbus.String(u'ServicesResolved'), dbus.String(u'UUIDs')]
Property changed [dbus.String(u'Alias'), dbus.String(u'Name')]
Property changed [dbus.String(u'Appearance'), dbus.String(u'Icon')]
--------------------------------------
org.bluez.GattCharacteristic1
Value Changed, I will do something here
dbus.Array([dbus.Byte(10)], signature=dbus.Signature('y'))
ERROR:dbus.service:Unable to append (None,) to message with signature v: <type 'exceptions.TypeError'>: Don't know which D-Bus type to use to encode type "NoneType"
--------------------------------------
org.bluez.GattCharacteristic1
Value Changed, I will do something here
dbus.Array([dbus.Byte(15)], signature=dbus.Signature('y'))
ERROR:dbus.service:Unable to append (None,) to message with signature v: <type 'exceptions.TypeError'>: Don't know which D-Bus type to use to encode type "NoneType"

If I just print something ("Value Changed, I will do something here" in this example)  in the property changed call back function, it can be successfully printed out. But when I try to do something else after the print statement, it has no response.

ukBaz, 2016-11-14:
Thanks @wuyj88 for sending over the more detailed log of what you are doing.

I think I've found the problem and it is in the localGATT file, class Characteristic and method WriteValue
The return value does not match the out_signature.

I've published my branch I've been experimenting in https://github.com/ukBaz/python-bluezero/tree/lightSwitch

The main change is to the signature of WriteValue:
https://github.com/ukBaz/python-bluezero/commit/e8029b5b53b6386c3c47369bbeaffa41043adee1#diff-fa4f64a0bbc9e93e1b48fb78d3b80673L353

And removing of the return statement
https://github.com/ukBaz/python-bluezero/commit/e8029b5b53b6386c3c47369bbeaffa41043adee1#diff-fa4f64a0bbc9e93e1b48fb78d3b80673L359

I don't think all the other changes are necessary that I've made plus I've broken the Travis CI testing so I will need to clean this up before merging into master but I wanted to get this to you to see if that was the fix. If this fixes it for you then I'll work on getting it into master.

wuyj88, 2016-11-15:
It works now, I appreciate.

First time for me to use dbus, it seems I need to get deeper into it.

Thank you for the contribution and working around!

ukBaz, 2016-11-15:
Thank you for letting me know this is working now.

closed 2016-11-15 by ukBaz