ukBaz/python-bluezero #108: Typo in method name _pin_pwn_control

ukBaz, 2016-12-01:
The method should be called `_pin_pwm_control` and not pw**_n_** as it is currently.

https://github.com/ukBaz/python-bluezero/blob/master/bluezero/microbit.py#L432

This would be a great simple pull request opportunity for anyone wanting to get involved with the project (or maybe just someone that wanted to practice doing a pull-request)

You will need to:
* Fork the repository.
* Make the fix.
* Submit a pull request to the project

Some tutorials that might be helpful are:
https://guides.github.com/activities/forking/
https://help.github.com/articles/fork-a-repo/
https://opensource.com/life/16/3/submit-github-pull-request

Kashyap12, 2016-12-01:
Hi @ukBaz , I would like to get involved, also I am quite new to open source world

ukBaz, 2016-12-01:
@Kashyap12  please go ahead and take a look to see if you can do a pull-request for this issue.

alexisgeoffrey, 2016-12-02:
Hey @ukBaz , since I see that no pull requests have been made, I'll give this one a try. I've never contributed to someone else's open source project before.

PR #110 (2016-12-02): fixed typos of method name _pin_pwm_control
fixes #108
closed 2016-12-02 by ukBaz