ukBaz/python-bluezero #111: dbus module missing the setup.py

benallard, 2016-12-02:
Obviously, bluezero needs dbus (adapter.py has an ```import dbus```). This should be in the dependency list in the setup.py so that installing bluezero pulls the right dependencies together.

benallard, 2016-12-02:
Looks like the correct dependency name is ``dbus-python``.

ukBaz, 2016-12-02:
Hello @benallard,
Can you tell me some more about this?
Up to now I've been using the instructions at: 
https://github.com/ukBaz/python-bluezero/blob/master/docs/install_bluez.rst
And the commands
```
sudo apt-get install python-dbus
sudo apt-get install python3-dbus
```
Which has caused me great pain on Travis-CI as apt-get will only work on certain versions of Python on Travis.
Is dbus-python a version that will install cleanly using pip? Does that mean I can use it on Travis-CI?

If this is true then you could have solved a big cause of pain for this project

benallard, 2016-12-02:
Unfortunately, not, this is not the solution to all issues. The ``dbus-python`` Pypi project would only (try to) install/compile the ``dbus`` module (provided the right development libraries are installed as well). Even with this installed the gi/gobject stuff is still missing ... Installing python3-dbus looks indeed like the easiest way to get all running.

Worth a mention in the README maybe ?

WayneKeenan, 2016-12-02:
Maybe just specifying an array of dependancies using the 'install_requires' key in setup.py will get you there?  (Sorry didn't twig sooner)

For example here's one of mine that uses it: https://github.com/WayneKeenan/python-vrzero/blob/master/setup.py

There is a caveat I've found...  modules installed this way remain in a zipped egg under Python's 'dist' folder. 
Usually not a problem, but it broke pi3d as that module tries to 'file open' resources in the module tree, which it expected to be expanded/unzipped fully.

benallard, 2016-12-02:
Na, you will not be able to put the gi/gobject stuff in the install_require array, as those are not available via Pypi.

I'm now convinced the best solution is to mention the mandatory additional installation pretty early, like on top of the README in a **Requirement** section.

ukBaz, 2021-01-01:
These modules normally come installed on the core systems that this library is targeted at. My concern with putting things in the readme that is unnecessary for the majority of uses is that it becomes overly wordy and confusing.

It has been sat here for 4 years without anyone adding to it. As a result, I'm going to close this.

closed 2021-01-01 by ukBaz