ukBaz/python-bluezero #112: Create an example peripheral to work with Science Journal app

ukBaz, 2016-12-09:
I've just stumbled across the Science Journal app from Google. It allows you to data log sensor data on your phone.
https://play.google.com/store/apps/details?id=com.google.android.apps.forscience.whistlepunk

The bit that caught my attention was that it can also do the logging for sensors over BLE. Their [GitHub repo](https://github.com/google/science-journal-arduino/blob/master/src/arduino/science-journal-arduino.cpp) has some Arduino examples. I would expect that this could be done with Bluezero also. The key bits of Arduino code with the BLE service and characteristic seem to be:
```
BLEService whistlepunkService("555a0001-0aaa-467a-9538-01f0652c74e8"); // create service
// Must be 20 char long to accomodate full-size messages.
const char *value = "                     ";
const char *config = "                     ";
BLECharacteristic valueCharacteristic( "555a0003-0aaa-467a-9538-01f0652c74e8", BLENotify, value);
BLECharacteristic configCharacteristic("555a0010-0aaa-467a-9538-01f0652c74e8", BLEWrite, config);
const unsigned short version = goosci_Version_Version_LATEST;

BLEUnsignedShortCharacteristic versionCharacteristic("555a0011-0aaa-467a-9538-01f0652c74e8", BLERead);
```

ukBaz, 2016-12-11:
This is a work-in-progress of what the understanding is of the Bluetooth profile required by Google Science Journal to be able to read Bluetooth (BLE) sensors

# Service

Field | Details
------|---------
Name | Science Journal
UUID | 555a0001-0aaa-467a-9538-01f0652c74e8

## Characteristic
Field | Details
------|---------
Name | Value
UUID | 555a0003-0aaa-467a-9538-01f0652c74e8
Flags | Read, Notify
Fields | utf8s
Summary | Sends the value of the sensor as a string (Max 20 characters)

## Characteristic
Field | Details
------|---------
Name | Config
UUID | 555a0010-0aaa-467a-9538-01f0652c74e8
Flags | Write
Fields | utf8s
Notes | Send instructions to the sensor as a string (Max 20 characters) ?What instructions does the app send?

## Characteristic
Field | Details
------|---------
Name | Version 
UUID | 555a0011-0aaa-467a-9538-01f0652c74e8
Flags | Read
Fields | ?unit8?
Notes | ?It is unclear to me what this should be set to but it looks like we might be on version 1 and it might be optional?

ukBaz, 2016-12-11:
I've been looking around the app files. The heavy lifting for the Bluetooth sensors seems to happen in the following file:
https://github.com/google/science-journal/blob/master/OpenScienceJournal/whistlepunk_library/src/main/java/com/google/android/apps/forscience/whistlepunk/sensors/BluetoothSensor.java#L204

The above link is to a comment which states:
`For Value (UUID:555a0003), it's a protobuf`
Which I think is referring to Google's language-neutral, platform-neutral, extensible mechanism for serializing structured data called [Protocol Buffers](https://github.com/google/protobuf.)

It is looking like the value is made up of a timestamp and value although the exact details are still a mystery at the time of writing.

ukBaz, 2016-12-12:
I asked a question in the [Google Science Journal Support Forum](https://productforums.google.com/d/msg/sciencejournal/tHcFeQrG0UA/cTFSPXV_CgAJ) about the Bluetooth Profile required and they have come back with "We do not have a BLE protocol layer that we are yet ready to support long-term" 

The solution is to build an Android application package (APK) that interfaces between Science Journal and the Bluetooth as detailed here:
https://github.com/google/science-journal/blob/master/api/ScienceJournalApi/README-api.md

Writing the APK seems outside the scope of this project so I'm closing this

closed 2016-12-12 by ukBaz

ukBaz, 2016-12-13:
Just wanted to capture here for future reference that the Bluetooth SIG has an adopted GATT specification for **Environmental Sensing Profile** and **Environmental Sensing Service**.
More information can be found at: https://www.bluetooth.com/specifications/adopted-specifications
This [PDF](https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=294797) and [XML](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.environmental_sensing.xml) is probably the most useful

Geoffysicist, 2020-10-04:
Did you get any further with this? Now that the science journal has gone over to arduino and they can link to the arduino nano sense, it should be possible to pursue this. I want to make a physics sensor based on the adafruit feather sense but would prefer to work in circuitpython because arduino is a pain.

ukBaz, 2020-10-04:
This idea got abandoned previously because there was not a documented API accessible from the Raspberry Pi. I believe this is still the situation. If you know of some documentation then please share and I can take a look.