ukBaz/python-bluezero #117: Advertisement include_tx_power setter typo

kellyjp, 2016-12-25:
The setter method for the include_tx_power property of the Advertisement class calls the "set" method  which doesn't exist.  It should instead call the "Set" method.

ukBaz, 2016-12-26:
Well spotted!

Feel free to do a pull request or I'll take a look once I can get back to my development environment in a week or so

ukBaz, 2016-12-31:
I've started to look at. Before making the simple fix I am taking a look at doing some tests first to stop such errors slipping through.

@mr499 : Have you looked at tests for advertising at all?

ukBaz, 2017-01-01:
I've just fixed the typo and done some manual testing so I can close this issue.

My investigations into getting something automated using Travis were unsuccessful.

If anyone has ideas how we can get better test coverage on this library with automated testing then please let me know.

closed 2017-01-01 by ukBaz