ukBaz/python-bluezero #120: BlueZero on a Pi 3

CosmaP, 2017-02-05:
Hi, 

I am fairly sure I have followed all the various instructions here but I am still getting

Traceback (most recent call last):
  File "microbit_button.py", line 15, in <module>
    from bluezero import constants
ImportError: No module named bluezero

Bluetooth status gives

● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled)
   Active: active (running) since Sun 2017-02-05 19:29:02 UTC; 8min ago
     Docs: man:bluetoothd(8)
 Main PID: 694 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─694 /usr/libexec/bluetooth/bluetoothd

Which means it's running (right?)

 bluetoothctl
[bluetooth]# [NEW] Controller B8:27:EB:AA:9F:FF CosmicPi05 [default]
[bluetooth]# version
Version 5.43

Bit of a newbie question.  Any ideas?

Is there a set of uptodate consolidated instructions?

Thanks

Cosma

ukBaz, 2017-02-05:
Hello @CosmaP

Thanks for opening the issue. Bear with me while I try to walk through your setup to see where the issue might be.
The BlueZ version and the bluetoothd all look good so you have got the foundation setup correctly.
My assumption is that you have downloaded the Bluezero library from the source on GitHub.
With that assumption I can get a similar error to you if the path to the downloaded library is not set. I can fix it by setting the `PYTHONPATH` environment variable.
For example:
```
$ python3 microbit_button.py F7:17:E4:09:C0:C6
Traceback (most recent call last):
  File "microbit_button.py", line 18, in <module>
    from bluezero import constants
ImportError: No module named 'bluezero'
$ export PYTHONPATH=~/python/python-bluezero
$ python3 microbit_button.py F7:17:E4:09:C0:C6
Button A
Button B
Bye bye!!!
```

Could this be the issue you are having?

Thanks,
Barry

CosmaP, 2017-02-05:
Many thanks for the help

 

I now get 

 

pi@CosmicPi05:~/Code/Cosmic/Robot $ sudo python microbit_button.py B6:27:EB:AA:9F:FF

Traceback (most recent call last):

  File "microbit_button.py", line 15, in <module>

    from bluezero import constants

ImportError: No module named bluezero

pi@CosmicPi05:~/Code/Cosmic/Robot $ cd ~/python

-bash: cd: /home/pi/python: No such file or directory

 

I do not have a ~/python/python-bluezero  folder

 

________________________________ 

Cosma Papouis

m: 07956 323938
Skype: Cosma.Papouis

e:  <mailto:cosma.papouis@uatechserv.com> cosma.papouis@uatechserv.com

CosmaP, 2017-02-05:
Possibly a bit cheeky.

 

But I can give you ssh access to this pi, if it will help.

 

Please send me a direct (non-public) email and I will send you the details.

 

Thx

 

________________________________ 

Cosma Papouis

Skype: Cosma.Papouis

e:  <mailto:cosma.papouis@uatechserv.com> cosma.papouis@uatechserv.com

ukBaz, 2017-02-05:
Hello @CosmaP ,

At what location do you have the `python-bluezero` directory that was created when you downloaded the Bluezero library?
It is that location that you need to put in the `export PYTHONPATH=<your path>/python-bluezero`

Give that a try.

Thanks,
Barry

CosmaP, 2017-02-05:
I left it in 

 

~/bluez/bluez-5.43

 

I have now copied to ~/python-bluezero

 

Have I cocked the install up?

 

________________________________ 

Cosma Papouis

ukBaz, 2017-02-05:
BlueZ and Bluezero are two separate libraries. Bluezero builds on top of the BlueZ DBus API. Having the python-bluzero directory anywhere on your system shouldn't be a problem as long as you tell Python where it is by using the PYTHONPATH setting.
Have you set PYTHONPATH?
Can you report back with:
1) How you downloaded Bluezero?
2) Where the python-bluezero directory is?
3) What you have PYTHONPATH set to?

Thanks,
Barry

CosmaP, 2017-02-05:
It sounds like I have not installed bluezero (I did follow the instructions I thought).

I will try again tomorrow.

Can you let me know how to install bluezero please.

Sorry for the newb mistake. I am not normally  this silly.

Thanks for the help so far.

Cosma

ukBaz, 2017-02-05:
Hello @CosmaP ,

You raise a good point. There are no specific instructions on how to install Bluezero. This is something that I'll look to update in the documentation of the library going forward.
The library is available from https://pypi.python.org/pypi/bluezero/ or https://github.com/ukBaz/python-bluezero and I've tried to follow standard methodologies for installs from those sites. However that is not helpful if Bluezero is your first library from these sites.

For now I'll give you a worked example of how to `clone` Bluezero from GitHub
```
cd
git clone https://github.com/ukBaz/python-bluezero.git
export PYTHONPATH=~/python-bluezero
python3 python-bluezero/examples/level10/microbit_button.py XX:XX:XX:XX:XX:XX
 ```
Where `XX:XX:XX:XX:XX:XX` is the address of your micro:bit running the Bluetooth profile.

Please remember the words under the Status section of the README.rst;  Bluezero is in early development so it has more than its fair share of rough edges. I tell you this to warn you that you are a pioneer so expect some issues. Reporting those issues are helpful to us in making it better for the people that follow after you. Also please keep in mind that this is a hobby project for all the contributors and response times to issues can vary depending on other commitments.

Hopefully we are over the worst and you will be up and running soon.

Regards,
Barry

CosmaP, 2017-02-06:
IT WORKS!!

Many thanks Barry, you are a hero.

Now I need to make it do something constructive.  Actually, it will be controlling a RasPi powered Dalek for Pi Wars.

Thanks again.

Cosma

PS.  Is there a reason why the export PYTHONPATH=~/python-bluezero does not persist?

ukBaz, 2017-02-06:
Thanks for letting me know it worked. Good news!

If you want the PYTHONPATH to be persistent then add it to your `~/.bashrc` file.

I'm going to close this for now. If you have other questions then please open another issue.

Good luck in Pi Wars. I'll keep my eye out for you...

closed 2017-02-06 by ukBaz