ukBaz/python-bluezero #121: How to implement secure characteristic?

maksimus1982, 2017-02-08:
Hello.
In my project I'd like to create GATT service with some characteristics.
And i want to make them secure (ask pairing to access them). Unsecure  characteristics works fine.
Is it Possible ?
I have tried some flags in localGATT.Characteristic function -   ['read', 'notify','secure'] ,  ['secure read', 'notify']...
No one works. 
Is it possible to create secure characteristics by your framework ?

Thank you

ukBaz, 2017-02-08:
Hello @maksimus1982 ,

Bluezero is a wrapper around the BlueZ DBus API. Bluezero is an attempt to reduce the learning curve of getting up and running but should not block any functionality so if the BlueZ DBus API can do it then we can make Bluezero do it.

Currently there has been no work on Bluezero to simplify the handling of secure characteristics so there is nothing I can point you towards within this repo.
However in the BlueZ [GATT API](git.kernel.org/cgit/bluetooth/bluez.git/tree/doc/gatt-api.txt) documentations we can see the following flags are used for secure characteristics:
- "secure-read" (Server only)
- "secure-write" (Server only)

There is also a secure characteristic [example ](git.kernel.org/cgit/bluetooth/bluez.git/tree/test/example-gatt-server#n565) in the BlueZ test directory.

The piece that is unclear to me is if you have to register an agent as described in the [Agent API](git.kernel.org/cgit/bluetooth/bluez.git/tree/doc/agent-api.txt)

I am happy to colobarate on trying to work this out as it will be something that we will want to add into Bluezero at some point so please ask more questions here.
You might also want to consider using the BlueZ mailing list. More details at http://www.bluez.org/development/lists/

jackdneilson, 2018-08-29:
Just a comment here to show interest, I've managed to hack together something using bluezero as a BLE peripheral and then managed permissions / bonding using hciconfig. Is there any chance of pairing functionality being added to this project? It would certainly be helpful in my use case. If not, can you suggest areas to look at to implement?

ukBaz, 2018-08-29:
Hello @jackdneilson !  Welcome to Bluezero. Good to hear the library has been of some help to you.

The first thing I should mention is that hciconfig has been [deprecated](https://www.spinics.net/lists/linux-bluetooth/msg69239.html). As a result I have not spent any time looking at it. I use `bluetoothctl` for the type of tasks I think you are referring to.

As I said previously, there are a wide variety of ways this kind of functionality could be used and it remains unclear to me how we would best fit this in the library such that it eased the learning curve.

It might be useful for the conversation if you could share the steps you are doing in `hciconfig`. We could then look to see how we would replicate them with the [Agent API](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/agent-api.txt)

It would also be useful to know which of the following values from the [Agent API ](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/agent-api.txt#n33) best reflects the input and output capabilities of your device:
"DisplayOnly", 
"DisplayYesNo",
"KeyboardOnly",
"NoInputNoOutput"
"KeyboardDisplay" 

There is also a discussion currently happening on the BlueZ mailing list with the subject ["bt "server" how to configure requiring passkey from connecting clients"](https://www.spinics.net/lists/linux-bluetooth/msg76910.html)
My reading is that it has not reached a conclusion yet but may be worth seeing if there are any clues.

Something else that might be useful for the discussion is the [Simple Agent](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/test/simple-agent) example in the BlueZ tree.
It is an example with      `capability = "KeyboardDisplay"`

owenber123, 2021-01-15:
Few years later I guess but was this ever worked on.  I'm looking for bonding support at security mode 1 level 2, Just Works.  I have just started on this project and would be happy to contribute.  Any previous work or recommendations would be helpful.  Thanks.

WayneKeenan, 2021-01-15:
The bonding works at the bluez level, using bluetoothctl... 

I’m using bluezero with micro:bits set up for ‘pairing just works’ 

That’s good for me at device level, is that good enough for your needs?

ukBaz, 2021-01-16:
@owenber123 there hasn't been a lot done on the pairing front for either the central or peripheral role. For the central role I have always steered people to using `bluetoothctl` because for a one-off provisioning step this seems like the most efficient thing.

Up until now the peripheral implementation inside of Bluezero has not had much attention. I am working in a separate branch looking to improve things with the peripheral but am not working on pairing currently.

I have just done a quick experiment with the `cpu_temperature` example where I changed the flag from `read` to `encrypt-read` and this prompted a pairing event to happen before I could read values. This was using `bluetoothctl` as the agent rather than registering one in the application.

I also set the agent to `NoInputNoOutput` in `bluetoothctl` and that seemed to let me bond without entering a code. Is this the same as `Just Works`?

Not sure how to easily set pairing capability to `NoInputNoOutput` in the Python code. The BlueZ mgmt API seems to offer easy access:
https://github.com/bluez/bluez/blob/b31e509eaf7603126a2d0f10779e267af34f6e88/doc/mgmt-api.txt#L1023-L1053

But because of the the following Python bug it is not easily available from Python:
https://bugs.python.org/issue36132

Don't really want to have the whole of scapy as a dependency just to get a Bluetooth user socket. If someone knows how to efficiently implement the user socket to access BTMGMT API just in Python that would be fantastic.

owenber123, 2021-02-11:
This was all very helpful thank you.  I was able to get everything working, just need to automate the accept pairing pop up that happens because of the bonding and implement some encryption.  Thanks again.

ukBaz, 2021-02-11:
OK, I'm going to close this issue because I think we have proved that it a question of setting the flags correctly and making sure the device is bonded.

The agent part of this should probably go under a different ticket  for people that don't want to use `bluetoothctl`.

I am sure there are a number of variations on what needs to be implement from the Agent API depending on the where this is moved.
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/agent-api.txt

However, the basic idea is something along the lines of:

```python
class Agent(dbus.service.Object):

    @dbus.service.method(AGENT_IFACE,
                         in_signature='o', out_signature='s')
    def RequestPinCode(self, device):
        print(f'RequestPinCode {device}')
        return '0000'

agent = Agent(bus, AGENT_PATH)
agnt_mngr = dbus.Interface(bus.get_object(BUS_NAME, AGNT_MNGR_PATH),
                           AGNT_MNGR_IFACE)
agnt_mngr.RegisterAgent(AGENT_PATH, CAPABILITY)
```

closed 2021-02-11 by ukBaz