ukBaz/python-bluezero #127: "Failed to register advertisement" when running "cpu_temperature.py"

JiaGuoWMU, 2017-05-12:
@ukBaz I installed Bluez 5.43 on Raspberry Pi 3 following the instructions. There are two issues happening. 

First, the status shown below has the error messages (last three lines):
```
May 11 23:06:45 raspberrypi bluetoothd[6575]: Bluetooth daemon 5.43
May 11 23:06:45 raspberrypi systemd[1]: Started Bluetooth service.
May 11 23:06:45 raspberrypi bluetoothd[6575]: Bluetooth management interface 1.9 initialized
May 11 23:06:45 raspberrypi bluetoothd[6575]: Failed to obtain handles for "Service Changed" characteristic
May 11 23:06:45 raspberrypi bluetoothd[6575]: Not enough free handles to register service
May 11 23:06:45 raspberrypi bluetoothd[6575]: gatt-example-adapter-driver: Input/output error (5)
```

Second, after I run "cpu_temperature.py" and connected the Raspberry Pi with a central (e.g., smartphone app), everything works correctly. However, once I stoped the python script and tries to run it again, then I get the following error message:
```
CPU temperature is 51.5C
b'\x1e\x14'
Failed to register advertisement: org.bluez.Error.Failed: Failed to register advertisement
```

Any idea how I could fix these issues? Appreciate it.

ukBaz, 2017-05-12:
Hi @JiaGuoWMU ,

I took a quick look at this and I think all the errors you are seeing are from when you restart the cpu_temperature.py script. 
The failure to register the advertisement is because the advertisement wasn't unregistered when the cpu_temperature script was stopped.
One of my goals with the example scripts is to keep them as simple as possible to help aid understanding. I find lots of error checking (while good for production) isn't that helpful when people are trying to learn.
The other comment I would make is that usually when people are making their board/RPi3 a peripheral they are not very often stopping and starting the script.
During development (when there is likely to be lots of stooping and starting) I recommend people restart the bluetoothd anyway so that it is starting in a known state. Basically I am saying that it is unlikely I will update this example.

If I was looking to make this script more robust I would look to put the [pi_cpu_monitor.start_bt()](https://github.com/ukBaz/python-bluezero/blob/master/examples/level100/cpu_temperature.py#L148) statement in a `try:except:` so that when the script is interrupted you can clean things up.
To clean things up I think you need to do an [UnregisterAdvertisement](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/advertising-api.txt#n99) and to stop the mainloop. Maybe creating a `stop_bt` method in the ble class would be a good way to do this. Although I'm sure there could be a few ways of doing it. Does that help?

I would also like to point you to issue #126 to highlight that it is likely Bluezero will move away from using python-dbus to using pydbus. This is another reason why I am reluctant to invest too much time in improving the current examples.
Development of Bluezero has stalled/slowed while a couple of issues with pydbus are worked out.

JiaGuoWMU, 2017-05-18:
@ukBaz Hi Barry, thanks for your prompt response. It is a good idea to use a `try:except:` and add `UnregisterAdvertisement`. 

Besides, I also found out the issue is partly because of the version of the bluez. I updated from v5.43 to v5.45, and the issue is solved.

ukBaz, 2017-05-19:
Really appreciate you coming back and adding what you've learnt.

Sounds like I can close this issue for now.

closed 2017-05-19 by ukBaz