ukBaz/python-bluezero #129: Unable to set arguments ('/ukBaz/bluezero/advertisement0000', {})

vbisbest, 2017-05-21:
Getting this error.  Raspberry Pi is up to date and using bluez 5.45.

ERROR:dbus.connection:Unable to set arguments ('/ukBaz/bluezero/advertisement0000', {}) according to signature None: <class 'ValueError'>: Unable to guess signature from an empty dict
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 604, in msg_reply_handler
    reply_handler(*message.get_args_list(**get_args_opts))
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 401, in _introspect_reply_handler
    self._introspect_execute_queue()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 387, in _introspect_execute_queue
    proxy_method(*args, **keywords)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 137, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 584, in call_async
    message.append(signature=signature, *args)
ValueError: Unable to guess signature from an empty dict

ukBaz, 2017-05-21:
I think you are going to have to give me a bit more information before I can help you.

It looks like you are trying to `RegisterAdvertisement` but it is not happy with something in the interface `org.bluez.LEAdvertisement1`.

The place I would look at first is the format of the values for the `LEAdvertisement1` properties.

I would also recommend doing a `sudo service bluetooth restart` before trying to run your script to flush out anything that is hanging around from a previous run

vbisbest, 2017-05-21:
Indeed I did.  When I upgraded bluez, it overwrite the bluetooth conf file.  I had to add -E to the bluetoothd line.  Thank you.

closed 2017-05-21 by vbisbest