ukBaz/python-bluezero #131: raspberry pi dbus.exceptions.DBusException: org.freedesktop.DBus.Error.InvalidArgs: No such property 'ServicesResolved'

RaspM, 2017-06-16:
pi@raspberrypi:~ $ sudo python3 /home/pi/python-bluezero/microbit_button.py E4:BE:8D:F3:BB:B4
Traceback (most recent call last):
  File "/home/pi/python-bluezero/microbit_button.py", line 188, in <module>
    central(str(args.address))
  File "/home/pi/python-bluezero/microbit_button.py", line 130, in central
    ubit.connect()
  File "/home/pi/python-bluezero/microbit_button.py", line 63, in connect
    'ServicesResolved'):
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.InvalidArgs: No such property 'ServicesResolved'

my raspberry pi and micro:bit is paired successfully   throughout, so i am wondering what is the issue .

[bluetooth]# paired-devices
Device E4:BE:8D:F3:BB:B4 BBC micro:bit

any help extended is appreciated. Thank you.

ukBaz, 2017-06-16:
Thanks for looking at Bluezero.

I am going to need some more information before I can give you any sensible suggestions.

Can you run `bluetoothctl` on the command line. Once using `bluetoothctl` can you run the following commands:
`version`
`show`
`connect E4:BE:8D:F3:BB:B4`

And post the output from that to this issue.

Thanks,
Barry

RaspM, 2017-06-19:
[image]
[image]
sorry foe the late reply.

ukBaz, 2017-06-19:
Hello,

Thanks for posting the requested information.
I can see the version being reported as 5.23. 
Your original error message is correct, in that version of BlueZ bluetoothd there was no `ServicesResolved` property. 
In version 5.23 the Python API was not at production status. You will need to upgrade to BlueZ version 5.43 as there have been a number of corrections to the API to get it into production status.

I have put some instructions together on how to upgrade BlueZ at:
https://github.com/ukBaz/python-bluezero/blob/master/docs/install_bluez.rst

I am going to close this issue as it is the version that is the fix. If you have other problems then please open a new issue.

Regards,
Barry

closed 2017-06-19 by RaspM