ukBaz/python-bluezero #132: Notify when writing to characteristic

SimoneCarnio, 2017-06-19:
I need to emulate a peripheral with my rpi, so I've installed a demo board and written some code starting from blinkt_ble.py. 
To test I'm using an iOS app. I can enable notification on my characteristic and I can write on it. 
If I change value from code my iOS app receive the notification,  if I write to characteristic from my iOS app I get notified of the new value.
Now I need to do something with this value on python side. 
I've tried to add a callback to my characteristic and after a lot of tries I've understood how to receive bytes when a write a value on it.
Following your example, I've created a class to receive the callback and associated with something like this:

bleIstance.myCharacteristic.add_call_back(CBIstance.receivingFunction).
Now inside "receivingFunction" I can find my bytes with "bytes = args[1]["Value"]"

Everything perfect, but in this way my iOS doesn't receive any notification so I cannot update the value on screen. 
I've checked the sourcecode but I'm not so experienced in python/dbus/ble, I suppose that with this callback I'm preventing the dbus notification to be sent.
If I comment out the callback all works normally.

Any idea on how to fix this?
    
Now I'm updating bluez to 5.45 to see if it fix something (also to fix #127 )

ukBaz, 2017-06-19:
Hello Simone,

Notifications are one of the tricky bits especially with python-dbus (and one of the reasons I'm trying to switch across to using pydbus).

In short, the example is over-writing the [propertiesChanged](https://github.com/ukBaz/python-bluezero/blob/6f738308544229852333b7ba08470ecb07159e36/bluezero/localGATT.py#L348) method to emit a DBus signal.

You might also find the [light_switch.py](https://github.com/ukBaz/python-bluezero/blob/58e5567a1e04affe321514e4a4187bc17e9fd4a6/examples/level100/light_switch.py) example useful as that emits a notification from within Python when the button is pressed.

The other example is [cpu_temperature.py ](https://github.com/ukBaz/python-bluezero/blob/58e5567a1e04affe321514e4a4187bc17e9fd4a6/examples/level100/cpu_temperature.py) example which has less indirection within it and uses a timer to emit a notification. The [temperature_cb](https://github.com/ukBaz/python-bluezero/blob/58e5567a1e04affe321514e4a4187bc17e9fd4a6/examples/level100/cpu_temperature.py#L52) uses the propertiesChanged method directly.

The other piece that I've found important sometimes when doing notification callbacks is to return True at the end of the callback if you want to carry on notifying.

Hopefully that helps you move in the right direction.

Regards,
Barry

SimoneCarnio, 2017-06-19:
I've just fucked up my system trying to install bluez 5.45 and I'm rebuilding my test distro, so I' cant try right now.
In cpu_temperature example, the notification is triggered [here](https://github.com/ukBaz/python-bluezero/blob/58e5567a1e04affe321514e4a4187bc17e9fd4a6/examples/level100/cpu_temperature.py#L59), right?   So, in my call back if call that I should notify my iOS app that something has changed?  Thank you for your support, I will try in the next hour.

ukBaz, 2017-06-20:
Yes it is the propertiesChanged that emits the notification if you have notify on.

propertiesChanged is a DBus 'thing'. More information is available at:
https://dbus.freedesktop.org/doc/dbus-specification.html

ukBaz, 2017-08-13:
Hi @SimoneCarnio ,

There has been no activity on this for a while so I am guessing you have got it working.
If you are still having issues then please re-open.

Thanks,
Barry

closed 2017-08-13 by ukBaz