ukBaz/python-bluezero #133: Using HCI connection to Linux kernel

ukBaz, 2017-07-26:
During the various discussions on #126 it was raised about experimenting with using the HCI interface directly to the kernel.

If this library takes that route I would like to see if this can be done within Python rather than depend on other libraries and C bindings.

```python
import socket
sock = socket.socket(socket.AF_BLUETOOTH, socket.SOCK_RAW, socket.BTPROTO_HCI)
sock.bind((0,))
my_bytes = bytearray()
my_bytes.append(0)
my_bytes.append(4)
sock.send(my_bytes)
```
Gives the following:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 22] Invalid argument
```

Any suggestions on what I'm doing wrong?

WayneKeenan, 2017-07-27:
think the way to go is to use struct.pack and fcntl.ioctl, e.g.

buffer = struct.pack("I", dev_id)
fcntl.ioctl(sock.fileno(), HCIDEVUP, buffer)

My test case gets past the data format but is hitting a complaint from inside the kernel:

`IOError: [Errno 25] Inappropriate ioctl for device`

My experiment is [here](https://gist.github.com/WayneKeenan/3ff68e7e72df18bdaeb82a91d2c68d67)

I think running as root might be required.   To circumvent that requirement Noble/Bleno use an external binary which has a ACL applied to the binary, that is something not possible for a 'script', AFAIK.

aversion to running a python bluezero daemon as root?

WayneKeenan, 2017-07-27:
The reason for the failure is because the HCI commands passed to ioctl need to be converted from the 'bare' number to a Linux ioctl encoding, e.g. http://code.activestate.com/recipes/578225-linux-ioctl-numbers-in-python/

I've had a quick try, doesn't work as-is, needs a bit more massaging, will take a look later.

WayneKeenan, 2017-07-27:
added a [noddy class in a gist](https://gist.github.com/WayneKeenan/bcaabcc6e7f2cd7cd38fbd2b0b916782) that the test just simply bounces the `hci0` interface.  

Needs to be run with sudo.

WayneKeenan, 2017-07-27:
This is the 'call stack' with links to the functions:

START		[enable_le_scan]( https://github.com/colin-guyon/py-bluetooth-utils/blob/master/bluetooth_utils.py#L161)
calls:   	[bt_hci_send_cmd](https://github.com/karulis/pybluez/blob/gattlib/bluez/btmodule.c#L2014					)   a.k.a `import bluetooth._bluetooth(.so) as bluez`
calls:   	[hci_send_cmd ](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/lib/hci.c#n1090)
calls:	[writev (POSIX/GLIBC)]( https://linux.die.net/man/3/writev)

WayneKeenan, 2017-07-28:
I think in my next experiment I'm going to Pythonise an example and supporting subset of the NodeJS BLE library [BluetoothHCISocket](https://github.com/sandeepmistry/node-bluetooth-hci-socket) by @sandeepmistry

WayneKeenan, 2017-07-29:
I have implemented the experiment and have LE Scanning working, please see https://github.com/TheBubbleworks/python-hcipy

WayneKeenan, 2017-07-29:
I meant to add that the prototype demonstrates 3 key aspects of using socket io for HCI:

1. sending HCI Commands (e.g. HCIDEVUP)
2. decoding HCI responses (e.g. HCIDEVINFO, EVT_LE_META_EVENT)
3. handling unsolicited HCI events (e.g. EVT_LE_META_EVENT for LE Scanning) 

I've also kept in mind the need to do automated testing so the core class has been written with future automated tests in mind, comments in the code.

mr499, 2017-07-29:
That's a great demonstration - I spent quite some time banging my head against a brick wall with a prototype similar to Barry's opener, without knowing about the IOCTL numbers (I was getting a "file descriptor in a bad state" type error from `fcntl.ioctl`.  I'm glad Barry asked the question and that there is a purely pythonic way to do HCI!

WayneKeenan, 2017-07-29:
I have just added the port of the LE advertisement test, https://github.com/TheBubbleworks/python-hcipy/blob/master/le_advertisement_test.py

WayneKeenan, 2017-08-03:
There is now a Connection and a PhysicalWeb Beacon example.  

Scripts can also can run as  a non-root user, please check out the [README](https://github.com/TheBubbleworks/python-hcipy) for more info.

I'm keen to understand the performance benefits of the approach, @ukBaz please could you outline the structure of a specific test scenario you had in mind? (it was beacon related I think)

ukBaz, 2017-08-03:
Good work @waynekeenan!

There are two performance issues that are a concern with DBus
1) continuously scanning for devices. E.g. if you want to constantly scan for nearby beacons. This was discussed on the Bluez mailing list 
http://marc.info/?l=linux-bluetooth&m=148123125327634&w=2
There is no way to do this with the DBus API currently
2) Throughput with read/write/notify of GATT characteristics. This was the issue raised on the Adafruit BLE library that was linked to on the Bluezero retrospective. 
New DBus API of AcquireNotify and AcquireWrite seems to be trying to address this. 
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/gatt-api.txt#n117

Does that help?

kaidokert, 2017-08-29:
Just as a comment, here is yet another half finished pile of code that kind of speaks HCI ( written for TI BLE embedded configurations ) 

https://github.com/mzxu/pyble/tree/master/ble

Assumes a serial port, but i've verified it to somewhat work with plain character device too for basic functions. At least the constant tables might be useful for anyone trying direct HCI interface implementation

ukBaz, 2017-08-29:
Thanks @kaidokert .

We should add this to the list of libraries on #126 

Maybe we should start a wiki with all these libraries listed.

kaidokert, 2017-08-29:
Not to flood, but i came across this too https://github.com/Jumperr-labs/hci-protocol . Built on top of https://en.wikipedia.org/wiki/Construct_(python_library) which provides nice self-documenting code.

Jumperr-labs also has python-gatt-server and jumper-ble-logger repos that use this implementation.

ukBaz, 2017-08-29:
Thanks @kaidokert .

I've started a wiki to track these libraries and make it easier for people to update directly:
https://github.com/ukBaz/python-bluezero/wiki

ukBaz, 2017-11-28:
Closing this and following the Python HCI sockets work happening elsewhere

closed 2017-11-28 by ukBaz