ukBaz/python-bluezero #134: How to implement micro:bit event service

ferrygun, 2017-08-09:
Do you have any sample code how to implement bluetooth event service on micro:bit? 

I have added these functions with the necessary UUID for the event service (event ID 100, event value 0), but it didn't work.

`ef event_config(self):
        self.event_client_req_characteristic_iface.WriteValue([100, 0], ())

    def _event_notify_cb(self):
        print('Event subscribed!!!')
        return 1

    def subscribe_event(self, user_callback):
        btn_obj = tools.get_dbus_obj(constants.BLUEZ_SERVICE_NAME,
                                     self.event_characteristic_path)
        btn_iface = tools.get_dbus_iface(constants.DBUS_PROP_IFACE,
                                         btn_obj)
        btn_iface.connect_to_signal('PropertiesChanged',
                                    user_callback)
        btn_obj.StartNotify(reply_handler=self._event_notify_cb,
                            error_handler=tools.generic_error_cb,
                            dbus_interface=constants.GATT_CHRC_IFACE)`

closed 2017-08-09 by ferrygun

ukBaz, 2017-08-09:
Hello @ferrygun ,

I see you have closed this issue. Does that mean you have solved the problem?
I'd be interested to hear how you did it and if anything could be made easier.

Thanks,
Barry

ferrygun, 2017-08-09:
hello,
Yes, I managed to resolve. I registered the bluetooth event service. Here is the modified code on my repo:
https://github.com/ferrygun/micro-bit-googleassistant/blob/master/bluezero/microbit.py