ukBaz/python-bluezero #141: Raspberry Pi <-> micro:bit workshop

ukBaz, 2017-08-12:
With Debian stretch release for Raspbian expected later this year it would be good to get a workshop in place to show how Raspberry Pi as a central device can interact with with a micro:bit as a peripheral.

The plan would be that the micro:bit would have the Bitty Software Bluetooth image on them 
http://www.bittysoftware.com/downloads.html#bitty_blue

The workshop would then take people through:
 - pairing
 - Sending text to the micro:bit
 - Receiving button presses

The additional challenge in the workshop would be to take button presses and accelerometer readings to draw [Python Turtle](https://docs.python.org/3.6/library/turtle.html) graphics

ukBaz, 2017-08-26:
A first draft has been posted at:
https://ukbaz.github.io/howto/ubit_workshop.html

johngouk, 2020-11-10:
I have tried the examples, they work fine. Just because it's interesting, I've tried using callbacks via Microbit.subscribe_button_a to detect button pushes. Strangely, the dBus code is calling my callback, not the Microbit callback. I've copied/modified the code in Microbit._decode_btn_a and then I'm able to pick up the button state value. 

Any ideas?? I'm happy to do that, but it seems a bit mad when the Microbit methods are supposed to do that...