ukBaz/python-bluezero #142: Investigate the BlueZ virtual adapter functionality

ukBaz, 2017-08-12:
BlueZ have the ability to create virtual adapters using the _undocumented_ btvirt tool.

I have had issues getting it to run and have asked the developer for some documentation.

The basic gist of it is you create two virtual adapters to form each end of the Bluetooth connection.
```
$ ./emulator/btvirt -h
btvirt - Bluetooth emulator
Usage:
	btvirt [options]
options:
	-S                    Create local serial port
	-s                    Create local server sockets
	-l [num]              Number of local controllers
	-L                    Create LE only controller
	-B                    Create BR/EDR only controller
	-A                    Create AMP controller
	-h, --help            Show help options
```

However I seem to be getting an error when I try:
```
$ sudo ./emulator/btvirt -l 2
Bluetooth emulator ver 5.43
Failed to open Virtual HCI device
```

AhmedHussein534, 2018-02-01:
you are using a wrong command the right one is $sudo ./emulator/btvirt -l2

ukBaz, 2018-02-01:
Ah! Makes sense.  I'll give that a try. 

@AhmedHussein534 , Do you have experience of using btvirt for doing automated testing?

AhmedHussein534, 2018-02-01:
no i started using it today

ukBaz, 2018-02-01:
@AhmedHussein534,

I tried it as you suggested but still not working for me:

```unix
pi@RPi3:~/bluez/bluez-5.43 $ sudo ./emulator/btvirt -l2
Bluetooth emulator ver 5.43
Failed to open Virtual HCI device
```

What version are you using?
Is there anything you did special?

If I select BR/ER only then it works:
```unix
pi@RPi3:~/bluez/bluez-5.43 $ sudo emulator/btvirt -B -l2
Bluetooth emulator ver 5.43
```
Then in another terminal I can see the extra adapters:
```unix
pi@RPi3:~ $ bluetoothctl list
[NEW] Controller 00:AA:01:01:00:24 RPi3 #3 [default]
[NEW] Controller 00:AA:01:00:00:23 RPi3 #2 
[NEW] Controller B8:27:EB:22:57:E0 RPi3 
```

AhmedHussein534, 2018-02-02:
No, I didn't do anything special, I only cloned  this
https://github.com/hadess/bluez

mr499, 2018-02-03:
`sudo` ./emulator/btvirt -l2` works for me (at least I can see an adapter in `bluetoothctl` in another terminal). @ukBaz to get `btvirt` I had to add `--enable-testing` to the configure options when building the Bluez source.

ukBaz, 2018-02-03:
Thanks @mr499. `--enable-testing` has the btvirt building now for 5.48. It is not required for 5.43.

I've upgraded my system to 5.48 using the following config:
```unix
./configure --prefix=/usr --mandir=/usr/share/man --sysconfdir=/etc --localstatedir=/var --enable-experimental --enable-maintainer-mode --enable-testing
```
This is still showing me the issue
```unix
pi@RPi3:~ $ sudo bluez/bluez-5.48/emulator/btvirt -l2
Bluetooth emulator ver 5.48
Failed to open Virtual HCI device
```

The status of various things are:
```unix
pi@RPi3:~ $ sudo service bluetooth status
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2018-02-03 19:20:46 UTC; 1min 0s ago
     Docs: man:bluetoothd(8)
 Main PID: 1020 (bluetoothd)
   Status: "Running"
   CGroup: /system.slice/bluetooth.service
           └─1020 /usr/libexec/bluetooth/bluetoothd --experimental

Feb 03 19:20:46 RPi3 systemd[1]: Starting Bluetooth service...
Feb 03 19:20:46 RPi3 bluetoothd[1020]: Bluetooth daemon 5.48
Feb 03 19:20:46 RPi3 systemd[1]: Started Bluetooth service.
Feb 03 19:20:46 RPi3 bluetoothd[1020]: Starting SDP server
Feb 03 19:20:46 RPi3 bluetoothd[1020]: Bluetooth management interface 1.14 initialized
Feb 03 19:20:46 RPi3 bluetoothd[1020]: Endpoint registered: sender=:1.22 path=/A2DP/SBC/Source/1
Feb 03 19:20:46 RPi3 bluetoothd[1020]: Endpoint registered: sender=:1.22 path=/A2DP/SBC/Sink/1
pi@RPi3:~ $ uname -a
Linux RPi3 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux
pi@RPi3:~ $ bluetoothctl -v
bluetoothctl: 5.48
```

`btmon` is not giving any clues:
```
pi@RPi3:~/bluez/bluez-5.48 $ sudo btmon
Bluetooth monitor ver 5.48
= Note: Linux version 4.9.59-v7+ (armv7l)                                              0.414158
= Note: Bluetooth subsystem version 2.22                                               0.414167
= New Index: B8:27:EB:22:57:E0 (Primary,UART,hci0)                              [hci0] 0.414172
@ MGMT Open: bluetoothd (privileged) version 1.14                             {0x0002} 0.414176
@ MGMT Open: bluetoothd (privileged) version 1.14                             {0x0001} 0.414180
@ MGMT Open: btmon (privileged) version 1.14                                  {0x0003} 0.414443
= New Index: 00:00:00:00:00:00 (Primary,Virtual,hci1)                           [hci1] 7.248199
= Open Index: 00:00:00:00:00:00                                                 [hci1] 7.248886
< HCI Command: Reset (0x03|0x0003) plen 0                                    #1 [hci1] 7.248938
= Close Index: 00:00:00:00:00:00                                               [hci1] 17.557466
= Delete Index: 00:00:00:00:00:00                                              [hci1] 17.557659
```

Not sure what else to try to identify where the issue is with my system...

ukBaz, 2018-02-05:
A little more investigating on this and the issue seems to be on the Raspberry Pi 3 hardware.

mr499, 2018-03-05:
I got it working on the Pi3 with a kernel rebuild - [setup & building](https://www.raspberrypi.org/documentation/linux/kernel/building.md), [configuration](https://www.raspberrypi.org/documentation/linux/kernel/configuring.md) - I used menuconfig, browsed to the crypto settings and enabled the userspace interfaces (not sure exactly which one is needed at this point), rebuilt the kernel and btvirt worked.  Next step is trying to build stuff as modules and figure out the specific module.

mr499, 2018-03-05:
As a follow-up, I rebuilt with "user-space for hash algorithms" (`CONFIG_CRYPTO_USER_API_HASH=m` in the `.config`) and that seems to work with `lsmod` now showing `algif_hash` referred to by `af_alg` (not sure what `af_alg` uses in a vanilla kernel).

ukBaz, 2021-01-01:
Doing an `apt-cache search bluez` on a Raspberry Pi has revealed the following package:
```none
bluez-test-tools - test tools of bluez
```
After installing with
```none
sudo apt install bluez-test-tools
```
The Bluetooth emulator is available:
```none
pi@SensePi:~ $ btvirt -h
btvirt - Bluetooth emulator
Usage:
        btvirt [options]
options:
        -S                    Create local serial port
        -s                    Create local server sockets
        -l [num]              Number of local controllers
        -L                    Create LE only controller
        -B                    Create BR/EDR only controller
        -A                    Create AMP controller
        -h, --help            Show help options
```
Doing `sudo btvirt -l2` does add two extra controllers:
```none
pi@SensePi:~ $ bluetoothctl list
Controller 00:AA:01:00:00:23 SensePi #2 [default]
Controller 00:AA:01:01:00:24 SensePi #3 
Controller B8:27:EB:22:57:E0 RPi_UART 
```

ukBaz, 2021-01-01:
Looking at the [latest specifications for GitHub-hosted runners](https://docs.github.com/en/free-pro-team@latest/actions/reference/specifications-for-github-hosted-runners) for GitHub Actions, it looks like `ubuntu-20.04` is that latest.

On a local machine running `Ubuntu 20.04.1 LTS` there is no `bluez-test-tools` package. There is `bluez-tests` but this does not appear to include `btvirt` looking at the list of files:
https://packages.ubuntu.com/groovy/i386/bluez-tests/filelist

Looking at the list for Debian based distros confirms that `btvirt` is included:
https://packages.debian.org/buster/i386/bluez-test-tools/filelist

ukBaz, 2021-01-03:
Built `btvirt` from source on Ubuntu. 

```none
wget http://www.kernel.org/pub/linux/bluetooth/bluez-5.54.tar.xz
tar xf bluez-5.54.tar.xz
./configure --prefix=/usr --mandir=/usr/share/man --sysconfdir=/etc --localstatedir=/var --enable-experimental --enable-maintainer-mode --enable-testing
make -j 4
```

> Don't do an install unless you want to change the version installed on your system (probably not what you want)

Emulator is now available for use:
```none
$ emulator/btvirt -h
btvirt - Bluetooth emulator
Usage:
	btvirt [options]
options:
	-S                    Create local serial port
	-s                    Create local server sockets
	-l [num]              Number of local controllers
	-L                    Create LE only controller
	-B                    Create BR/EDR only controller
	-A                    Create AMP controller
	-h, --help            Show help options
```

Using it:
```none
$ sudo emulator/btvirt -l2 &
[1] 25170
Bluetooth emulator ver 5.54
``
`bluetoothctl list` will show two new adapters available