ukBaz/python-bluezero #149: advertisement.py exposes mainloop. Is this correct?

ukBaz, 2017-08-25:
The example given in the advertisement is as follows:

    >>> from bluezero import tools
    >>> from bluezero import advertisement
    >>> beacon = advertisement.Advertisement(1, 'broadcast')
    >>> beacon.service_UUIDs = ['FEAA']
    >>> beacon.service_data =  {'FEAA': [0x10, 0x00, 0x00, 0x63, 0x73,
    >>>                                  0x72, 0x00, 0x61, 0x62, 0x6f,
    >>>                                  0x75, 0x74]}
    >>> ad_manager = advertisement.AdvertisingManager()
    >>> ad_manager.register_advertisement(beacon, {})
    >>> tools.start_mainloop()

This means that in a 'level 10' file you would need to know about mainloops which according to Bluezero's overview documentation 'It will not make reference to DBus or event loops.'

Maybe the API should be updated so the last line of the example would be:

    >>> ad_manager.start()

and then that starts the mainloop

jrl290, 2017-08-31:
`tools` doesn't have a `start_mainloop()` function nor does `AdvertisingManager` have a `start()` function

Please advise

ukBaz, 2017-08-31:
Ah! yes. the `start_mainloop()` has been moved to `dbus_tools`.

This is on the 'todo' list to fix by putting a `start()` in `AdvertisingManager`

Is that enough information to get you going?

If you fix it and want to create a PR then it will be happily accepted  :-)

jrl290, 2017-08-31:
This would be my very first python project, and I have never created a PR on github before, so I will not risk destroying this project with my inexperience.

But yes, pointing to `dbus_tools` worked fine. Thank you!

ukBaz, 2017-08-31:
Glad it worked!

I'll leave it on my todo list  :-)

ukBaz, 2017-09-03:
The commit 5947a2b6e31b2bccd0f24e87ac6e5487b787c419 closes this issue

closed 2017-09-03 by ukBaz