ukBaz/python-bluezero #157: dbus_tools.get_dbus_path returns None

schwa, 2017-09-10:
I'm trying to get some examples to work (modified level10 light switch code) on a Raspberry pi running latest raspbian. I believe I've set up bluezero's environment correctly (adapter_example,py runs fine). But when calling peripheral.Application() from my code I get an exception. Doing some minor tracing of the error it looks like dbus_tools.
_get_dbus_path2() returns None. It loops through the ifaces but nothing matches and it falls out of the loop and returns None.

Any help appreciated.

```
from bluezero import peripheral
from bluezero import adapter
from bluezero import dbus_tools

#peripheral.Application(adapter='/org/bluez/hci0')
#adapter.Adapter(adapter_addr='/org/bluez/hci0')
print dbus_tools.get_dbus_path(adapter='/org/bluez/hci0')
```

tyten, 2017-09-11:
You've got to use the actual address of the adapter and not the path. The current release seems really broken at the moment.

tyten, 2017-09-11:
Just look at this commit to see what the current state is. https://github.com/ukBaz/python-bluezero/commit/86715902baf1360dd8574186eee6caae6452ae10#diff-89737a96a9d04d54a81f2b1057411c71

tyten, 2017-09-11:
Here's an example cpu_temperature.py I was able to modify and get working. https://gist.github.com/tyten/0efb90d998c59d5926ca7f4e79054361

ukBaz, 2017-09-11:
@schwa : There has been a conscious decision to switch from using dbus path to adapter address. Using the dbus address would make it hard [if|when] we want to add support for other OSes that do not use dbus. This is why when calling "level 100" files it should not use dbus terminology so that OS specific things are restricted to within those few files.

@tyten : Thanks for finding the issue and posting a fix. Very much appreciated. Do you want to make that a PR?

> The current release seems really broken at the moment.

It would be great if you could give some examples of this. There are two main reasons that you might be seeing this:

1. the latest release had quite a few changes to it API's and that might be breaking your existing code. 
1. the test coverage is not very high and I may have missed places that needed updating 

If the reason is 1. then apologies for breaking things however I have done it with the best of intentions for the future of this library.
If it is reason 2. then please submit PR's or open issues where you are seeing problems.

I am also open to ideas of how to get the test coverage up on this library. My goal is to have all the examples be tested on Travis so that there can be higher confidence they are working.

ukBaz, 2017-09-11:
I am going to close this as I think we have resolved the problem in the title of this issue.

I have opened #159 to better reflect what needs to be done still

closed 2017-09-11 by ukBaz

tyten, 2017-09-14:
Thank you for doing that. Sorry it took me until now to read this and respond. I'll take a second stab at this and post what I find /submit a PR.