ukBaz/python-bluezero #158: DBUS permissions problems on Pi

mvellon, 2017-09-11:
I'm trying to run the cpu_temp example on a RPI Zero W, but I get an access denied error in connection.py telling me that the code is "not allowed to own the service 'ukBaz.bluezero" due to security policies in the configuration file.'  Note that I've looked at other bugs suggesting that I make changes to /dbus-1/system.d/bluetooth.conf, but it looks like my bluetooth.conf already has the necessary changes.  Additionally, I've added both "pi" and "root" to the bluetooth OS group, to no avail.

Any other suggestions?

ukBaz, 2017-09-11:
Are you talking about this?
https://github.com/ukBaz/python-bluezero/blob/master/docs/install_bluez.rst#creating-a-bluezero-peripheral
```
sudo cp examples/ukBaz.bluezero.conf /etc/dbus-1/system.d/.
```

Much of the peripheral functionality for BlueZ is still under the experimental flag so you need to enable that also.
This is documented at:
https://github.com/ukBaz/python-bluezero/blob/master/docs/install_bluez.rst#automatically-run-bluetoothd-with-experimental-mode

This all assumes you are using the Stretch version of Raspbian on the Zero W.

The CPU temperature example has an error in it as raised in #157 so you will need to fix that also.

Please let me know how you get on with this as it sounds like I need to make some updates to the documentation and example to make all of this clearer.

Thanks

mvellon, 2017-09-11:
Ah - I'd missed the changes to the bluetooth dbus configuration.  Let me play with this some more and I'll let you know how it goes.  Changing the dbus configuration got me past the permission error, but now I've got a different problem.

thanks

mvellon, 2017-09-11:
My RPI is running 5.23 of BlueZ which looks to be too old.  I'm running into different problems now which I'm assuming are due to not running 5.43.  I don't want to be modifying our base RPI images, so I'll put off more work on this until Raspbian gets an updated BlueZ.

Thanks for your help.

ukBaz, 2017-09-11:
OK, that's fine. I'll go ahead and close this issue as I think we have resolved the issue you raised.

FYI: 
The current version of Raspbian has BlueZ version 5.43.
More details available in this blog:
https://www.raspberrypi.org/blog/raspbian-stretch/
Including instructions on how to upgrade without wiping your system. They can be found in the section titled: **_How To Get Raspbian Stretch_**

closed 2017-09-11 by ukBaz