ukBaz/python-bluezero #163: Multiple Connections in Peripheral Mode

rac146, 2017-09-28:
Hey all,

I'm trying to figure out how to create a peripheral device that allows multiple connections. From what I've gathered, this should be possible w/ a Bluetooth 4.1+ adapter, although it seems to be dependent on the stack implementation. Currently I have an environment set up with a Raspberry Pi Zero W w/ BlueZ 5.47. The BT hardware in the RPI is the BCM43438 (4.1).

I can run the cpu_temperature GATT server example successfully, however it only allows 1 connection at a time. Anyone know if this limitation is related to the example, bluezero, or the BlueZ stack itself?

*When I ran the example, I was attempting 1 connection from Android and 1 from iPhone. I noticed for Android connections that the `/etc/bluetooth/main.conf` file should be switched to BLE only. I haven't tried this yet, but if my device was forcing a classic connection, I could see this being an issue.

ukBaz, 2017-09-28:
Hello @rac146 ,

Maybe I should just clarify that I am understanding the questions correctly.

In your example you want the Raspberry Pi Zero W to be in the peripheral role. You then have 2 devices in the Central role. An ios phone and an Android phone.

You have successfully connected from the ios phone and read the temperature from the RPi.

You have not been able to do the same from the Android Phone. Correct?
Is the Android phone able to see the advertisement from the RPi?
Is the behaviour that the Android phone connects and then instantly disconnect?

What application are you using on the phones to connect to the RPi?

ukBaz, 2017-09-28:
Thank you for clarifying.

Peripherals are generally optimized for connecting to a single central device and you will notice that BlueZ stops the RPi advertising once it has received a connection.
Normally with a peripheral, it has to disconnect from the central device before another one can connect.

A way to share information with multiple central devices is to encode the information in the advertisement data as the central devices do not have to connect to get the information.
This has got easier to do in BlueZ 5.47 as you can dynamically update the advertising information.

rac146, 2017-09-28:
Ya I realize my use case isn't going to be very common here. I basically want a host (RPi) to support multiple phone connections via BLE to send/receive small amounts of data. Since most phones these days provide good support for the Central role, I thought it would make the most sense to make the RPi a Peripheral.

Do you know of any way to force BlueZ to continue advertisement once a connection is established? Since the RPi Bluetooth adapter is 4.1, the hardware should support this less-common scenario.

I'll definitely look into the advertisement-only approach and see if it will fit my needs.

ukBaz, 2017-09-28:
I have had some time to run experiments and have got the same results as you.

The advertisement-only approach got enabled with this patch:
https://marc.info/?l=linux-bluetooth&m=150278513020056&w=2

The best place to ask about BlueZ as a peripheral to multiple centrals is on the BlueZ mailing list. More details at:
http://www.bluez.org/contact/

I would be interested to hear how you get on either with advertising-only or multiple connections so please report back.

For this ticket, I am going to close it for now as I am not sure Bluezero can help you at this moment.

closed 2017-09-28 by ukBaz

rac146, 2017-09-29:
Will do, appreciate the help!

chrisballinger, 2018-05-16:
@rac146 Any response on the BlueZ mailing list about supporting multiple centrals in peripheral mode?

rac146, 2018-05-17:
@chrisballinger I actually went ahead and ported my app to the nRF52832 platform by Nordic:
https://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF52832

The hardware is just way more suitable for this scenario, and supports up to 20 central connections! They have extensive API's for all kinds of things. If you're comfortable w/ C/C++, I would suggest checking out one of their dev boards!

chrisballinger, 2018-05-17:
@rac146 Oh neat! We are still prototyping but that seems like a great option for making real products :)

german-wings, 2019-07-12:
Hello,
Anything on "How to Connect multiple devices to Peripheral Server at once ?" I just started looking on this. @rac146