ukBaz/python-bluezero #171: Adapter() name & alias attributes have no effect?

eriknyquist, 2017-11-27:
I am using a raspberry pi 3. By default, the adapter alias is "raspberrypi", and when I create a GATTserver with

    sudo hcitool -i hci0 cmd 0x08 0x0008 1E 02 01 1A 1A FF 4C 00 02 15 E2 0A 39 F4 73 F5 4B C4 A1 2F 17 D1 AD 07 A9 61 00 00 00 00 C8 00

I can see the name "raspberrypi" on my phone (using the nRF Connect app from Nordic to connect).

Now, when I run your cpu_temperature.py example on the raspberry pi, it mostly works (i.e. I can see the peripheral address and connect on my phone), but the name does not appear, I just see the hardware address. I tried disconnecting/reconnecting several times with no success.

How can I get python-bluezero to show the correct alias to central devices (i.e. the one returned by localGATT.adapter.alias)?

Thanks,
Erik

ukBaz, 2017-11-28:
Hello Erik,

This is a very good point that you raise and the answer is buried in the details of the BlueZ DBus API.

Assuming the Raspberry Pi 3 is running the latest version of Raspbian then the default BlueZ version is 5.43 and so the Advertising API is as follows:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/advertising-api.txt?h=5.43

Since that version of BlueZ there has been an update to give more control over the local name that is advertised:
https://marc.info/?l=linux-bluetooth&m=150237199422018&w=2

So the latest API looks like:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/advertising-api.txt#n72

With the Bluezero library a decision was taken to focus on the version of BlueZ that ships with Raspbian/Debian so I have not investigated further with the new API.

In the past I have done some experiments with changing the name of the adapter globally in `/etc/bluetooth/main.conf`
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/main.conf#n5
with some mixed results.

Finally, for the very brave that are very comfortable with HCI commands, you can send the raw HCI instructions from Python to the Bluetooth adapter via sockets.

I would be interested to hear any updates on progress if you investigate these (or other) solutions.

Thanks,
Barry

ukBaz, 2017-11-28:
There is some overlap with #104

eriknyquist, 2017-11-28:
Thanks for the reply, yes, I tried changing the name in main.conf, but this makes no difference. The device still has no name when shown in a scan (not even the default peripheral name, "raspberrypi"), just the HW address.

Which version of BlueZ should be matched with the latest version of python-bluezero (i.e. master branch on this repository)?

I really want to use the new version of BlueZ, since the peripheral API is no longer marked as 'experimental'. Do you know of any good python libraries that wrap the BlueZ peripheral stack (the newest API)?

ukBaz, 2017-11-28:
Hi Erik,

> Which version of BlueZ should be matched with the latest version of python-bluezero (i.e. master branch on this repository)?

Bluezero is designed to work with BlueZ 5.43 as that is the version shipping by default with Raspbian/Debian. This is because the typical consumer of Bluezero does not want to rebuild BlueZ.

> I really want to use the new version of BlueZ, since the peripheral API is no longer marked as 'experimental'. Do you know of any good python libraries that wrap the BlueZ peripheral stack (the newest API)?

Sounds like you are more on than the _bleeding edge_ than most libraries are likely to be. 
You should just use the BlueZ DBus API directly. That is how BlueZ expect people to use it.
If you need some pointers then looking in the Bluezero files should give you some clues. There are also examples at https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/test

If you want to look for other libraries then I have a (not necessarily exhaustive) list at:
https://github.com/ukBaz/python-bluezero/wiki

Good luck!

I'll close this issue for now. Please re-open if necessary.

Regards,
Barry

closed 2017-11-28 by ukBaz