ukBaz/python-bluezero #175: Missing documentation ? Characteristic with string value

razerraz, 2018-01-04:
Hi,

It's more an help request than a bug report, but I don't found any clues for my problem in the documentation.

I'm trying to modify the cpu_temperature.py example, having a string as value instead of initial stuff I don't really understand.
My main change is the return value of ReadValue method : return dbus.ByteArray(reading.encode('utf-8'))

On the other side (self made android app), I got nothing.
I'm able to get this working using an esp32 µC (ie having characteristic with string type value)

Thanks in advance for your help

ukBaz, 2018-01-04:
Hi,

I think you are asking for help with changing the signed 16-bit integer (sint16) currently sent by the characteristic to a UTF-8 string. Is that correct?

There are a couple of points in the code that specify it is currently an sint16 characteristic:
1. The characteristics has a UUID of '2A6E' which specifies it as a [Temperature Characteristic](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.temperature.xml) which expected to be an sint16.
2. The [characteristic presentation descriptor](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.characteristic_presentation_format.xml) ('2904') for the characteristic is specified as an sint16, exponent of -2 and Celsius temperature value

I do not know if this is contributing to your issue as there is nothing in your request to suggest that you are getting any error messages.
You might want to have a look at the [Notes for getting debug information](http://bluezero.readthedocs.io/en/0.0.6/install_bluez.html#notes-for-getting-debug-information) section to see if there is any useful information being output by these to help us debug.

I will also have a look at creating something kind of text value example. I have been thinking about doing a [UART Service](https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html) similar to what is on the micro:bit

Regards,
Barry

ukBaz, 2018-01-04:
OK, it's not pretty but I have something that appears to be working. It is not the UART Service I was hoping as I have ran out of time this evening.

My experiment is below. Please let me know how you get on with your Android App. I have been testing with the [nRF Connect for Mobile](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) app

```python
# Standard modules
import random
import dbus
try:
    from gi.repository import GObject
except ImportError:
    import gobject as GObject

# Bluezero modules
from bluezero import constants
from bluezero import adapter
from bluezero import advertisement
from bluezero import localGATT
from bluezero import GATT

# constants
TEXT_SRVC = '6E408888-B5A3-F393-E0A9-E50E24DCCA9E'
TEXT_TX_CHRC = '6E409999-B5A3-F393-E0A9-E50E24DCCA9E'

def get_txt_value():
    text_values = ['Random words', 'Other stuff', 'Test value 2', 'Another sentence']
    return random.choice(text_values)

class TxChrc(localGATT.Characteristic):
    def __init__(self, service):
        localGATT.Characteristic.__init__(self,
                                          1,
                                          TEXT_TX_CHRC,
                                          service,
                                          [get_txt_value()],
                                          False,
                                          ['read', 'notify'])

    def value_cb(self):
        reading = get_txt_value()

        self.props[constants.GATT_CHRC_IFACE]['Value'] = reading.encode('utf-8')

        self.PropertiesChanged(constants.GATT_CHRC_IFACE,
                               {'Value': dbus.ByteArray(self.props[constants.GATT_CHRC_IFACE]['Value'])},
                               [])
        return self.props[constants.GATT_CHRC_IFACE]['Notifying']

    def _update_value(self):
        if not self.props[constants.GATT_CHRC_IFACE]['Notifying']:
            return

        print('Starting timer event')
        GObject.timeout_add(500, self.value_cb)

    def ReadValue(self, options):
        reading = get_txt_value()
        self.props[constants.GATT_CHRC_IFACE]['Value'] = reading.encode('utf-8')
        print('Value from ReadValue: {}'.format(reading))
        return dbus.ByteArray(self.props[constants.GATT_CHRC_IFACE]['Value'])

    def StartNotify(self):
        if self.props[constants.GATT_CHRC_IFACE]['Notifying']:
            print('Already notifying, nothing to do')
            return
        print('Notifying on')
        self.props[constants.GATT_CHRC_IFACE]['Notifying'] = True
        self._update_value()

    def StopNotify(self):
        if not self.props[constants.GATT_CHRC_IFACE]['Notifying']:
            print('Not notifying, nothing to do')
            return

        print('Notifying off')
        self.props[constants.GATT_CHRC_IFACE]['Notifying'] = False
        self._update_value()

class ble:
    def __init__(self):
        self.bus = dbus.SystemBus()
        self.app = localGATT.Application()
        self.srv = localGATT.Service(1, TEXT_SRVC, True)

        self.charc = TxChrc(self.srv)
        self.charc.service = self.srv.path

        self.app.add_managed_object(self.srv)
        self.app.add_managed_object(self.charc)

        self.srv_mng = GATT.GattManager(adapter.list_adapters()[0])
        self.srv_mng.register_application(self.app, {})

        self.dongle = adapter.Adapter(adapter.list_adapters()[0])
        advert = advertisement.Advertisement(1, 'peripheral')

        advert.service_UUIDs = [TEXT_SRVC]
        if not self.dongle.powered:
            self.dongle.powered = True
        ad_manager = advertisement.AdvertisingManager(self.dongle.address)
        ad_manager.register_advertisement(advert, {})

    def add_call_back(self, callback):
        self.charc.PropertiesChanged = callback

    def start_bt(self):
        self.app.start()

if __name__ == '__main__':
    pi_txt_service = ble()
    pi_txt_service.start_bt()
```

razerraz, 2018-01-05:
Thanks a lot ukBaz, I got it myself yesterday. Thé annoying thing is that I have nothing at all as output when something goes wrong during ReadValue method call. No exception, no callstrace... I simply had made a typo somewhere, something like obj.length() instead of len(obj) (too much java, i got sick), and nothing un value with nom clue at all of what was going wrong....
I guess it's dbus related, like my difficulty to understand tour code enough to have a good use of it. I have to learn dbus.

My other concern is about using glib as a scheduler/task manager un your exemple. It's a big dependency that may be replaced by core methods, asyncio or threads ?

Finaly, I made an AUR package of your module fit archlinux.

Cheers

ukBaz, 2018-01-05:
Thanks for the update. Very much appreciated. Good to hear you got it working.

You seem to have identified the weaknesses around Bluetooth on Linux :-)
The author of the dbus-python library admits that it has a number of issues: 

> Problems and alternatives
> 
> dbus-python might not be the best D-Bus binding for you to use. dbus-python does not follow the principle of “In the face of ambiguity, refuse the temptation to guess”, and can’t be changed to not do so without seriously breaking compatibility.
> 
> In addition, it uses libdbus (which has known problems with multi-threaded use) and attempts to be main-loop-agnostic (which means you have to select a suitable main loop for your application).

I've looked around at alternatives but they all seem to have issues. Maybe some type checking would help things.

The reason for going with GLib is that it matches the event loop used by BlueZ. It's a new area to me so if you have a better example then please let me know.

I am going to close this ticket as you have your example working. The bigger issues of python DBus bindings and event loops are covered elsewhere.

closed 2018-01-05 by ukBaz

razerraz, 2018-01-05:
As I see it, you have to deal with choices made upon by bluez, ie dbus/glib...
And by the way, weaknesses on linux are far from the messy android java ble implementation, the rain of callbacks You need to deal with, and the insane complexity of UI making google made.
I don't suffer with issues getting a gatt server working on linux, one way or other, just took the python way as usual when I have choices