ukBaz/python-bluezero #176: Device disconnect / out of range handling

PVince81, 2018-01-23:
I have a script that listens to events from a bluetooth device's characteristic. The device sends events periodically.

Whenever I move the device away from the computer, events stop arriving but also no exception is happening.

If technically possible through the DBus interface, I'd suggest the following:

1) Provide a callback/event for whenever the device is out of range or disconnects and let the developer decide to abort or reconnect

2) Maybe provide a flag to automatically reconnect in case of error

3) Option to abort main loop ?

PVince81, 2018-01-23:
Ouch, apparently it's worse than I thought: even Ctrl+C cannot abort the loop in case of disconnect:
```
  File "./test-bluezero.py", line 288, in <module>
    app.run()
  File "./test-bluezero.py", line 249, in run
    self.service.run()
  File "./test-bluezero.py", line 114, in run
    self.central.run()
  File "/usr/lib/python3.6/site-packages/bluezero/central.py", line 91, in run
    self.dongle.run()
  File "/usr/lib/python3.6/site-packages/bluezero/adapter.py", line 228, in run
    self.mainloop.run()
  File "/usr/lib/python3.6/site-packages/bluezero/async_tools.py", line 30, in run
    self.mainloop.run()
  File "/usr/lib64/python3.6/site-packages/gi/overrides/GLib.py", line 588, in run
    raise KeyboardInterrupt
```

I needed to kill the script from the CLI.

ukBaz, 2018-01-23:
Thanks for the feedback.

 I believe the issues you raise here are in an area of the library where there is a lot scope for improvement so it is good to get feedback on them.

To answer your individual suggestions:
1) Yes you are correct there does need to be something for when the device disconnects. DBus will emit a `PropertiesChanged` signal for the device interface when the status of the property `connected` changes i.e. goes from `True` to `False`. My initial thought is to have an `on_disconnect` that users can register a callback against.

2) Bluetooth does have some concept of automatically reconnecting if in range. I *think* it is to do with the `trusted` property. I will need to do some research. However if 1) is implemented that will also allows users to control re-connection if they wish.

3) There is a `quit` function on the mainloop. Does that not do what you want?
https://github.com/ukBaz/python-bluezero/blob/3d4e79828b3dc750e3c2a20a839db668dfaca768/bluezero/central.py#L93

PVince81, 2018-01-29:
I think 1 would already be a great addition. From what I observed it seems my watch doesn't reconnect, but it could also be due to not having a proper discovery handler. I need to study a bit on this. From my understanding it should be possible to somehow tell bluez that if it finds a device with a specific characteristic, it should automatically connect to it, like for bluetooth headsets. Maybe implementing a custom handler in a similar fashion (using the bluezero or DBus APIs), it would automatically reconnect.

Regarding ctrl+c handler, it seems to not work because of a Gtk bug: https://bugzilla.gnome.org/show_bug.cgi?id=622084#c35. It seems to be very recent so I'll see if it goes away with updates of the lib.

boardend, 2018-02-12:
I think i ran into the same Issue as @PVince81 At the moment there is no way to handle a "lost" client or to disconnect devices properly. 

Because of the lack of a disconnect-handler (proposed by @ukBaz as 1) )  I've implemented my own little watchdog which will detect when no BLE communication happend for some amount of time. After this timeout I try to reset bluezero that the micro:bit can be reconnected again. But i found no way to cleanup/reset bluezero completely. After the second connection the communication feels much slower, and I get multiple `Resolving GATT database for [UID]` while reconnecting. IMHO all registered callbacks etc. remain in the D-Bus, even when i discard  the microbit.Microbit instance completely.

My question is, how I can (re)use bluezero for multiple connections in the same thread/module?

Because I have a dedicated RaspberryPi 3 communicating to the micro:bit, I wouldn't care to fix things  with a crowbar. So I tried to restart the D-Bus and Bluetooth system services after a disconnect, reload the bluezero module, but I still get all those entries in the GATT database during the second connection attempt.

ukBaz, 2018-02-12:
Thanks @boardend .   Sounds like I need to up the priority of looking at this as you and @PVince81 have both raised this.

Let me make sure I understand correctly what you want here (or at least the first step I need to take).

The test I have in my mind is that when I turn off (take out of range) the peripheral (micro:bit) then a specified `on_disconnect` callback is triggered by Bluezero. That would then allow people to specify in that callback what should happen. i.e start scanning again or tidy up the current connection.

Is that OK with you both?

If anyone else is following this and has a view then this is your chance to raise any questions or concerns.

boardend, 2018-02-12:
Thanks for your quick response. Yes, such a `on_disconnect` callback would be the right place to handle the loss of a peripheral.

I'm not sure about the "tidy up" part yet. Perhaps I have overlooked something, but I couldn't find a Level 1/10 way of resetting bluezero, so that a new device instance can be created and connected.
Or maybe I'm just plain wrong and I should rather reuse the `bluezero.microbit.Microbit` instance and just reconnect without trying to quit the EventLoop and cleaning the whole D-Bus information/state?

PVince81, 2018-02-13:
I agree with @boardend on solution with `on_disconnect` 😄 

> After the second connection the communication feels much slower

I've raised a separate issue to discuss this to avoid mixing up topics: https://github.com/ukBaz/python-bluezero/issues/180

ukBaz, 2018-02-13:
I have had some time to look at this tonight. I don't have all the answers yet and it is not in a state to that it could be merged yet. However I hope it is enough to progress our conversation.
I have the `on_disconnect` branch that people can experiment with. This is the commit I've made this evening.
963f3673f6778f472cbd7fdb33d8e5fa650c2aa1

My testcase is this:
```python
from bluezero import microbit

ubit = microbit.Microbit(adapter_addr='B8:27:EB:22:57:E0',
                         device_addr='F1:55:90:65:29:DC')

print(ubit.ubit.dongle.powered)
if not ubit.ubit.dongle.powered:
    print('powering on')
    ubit.ubit.dongle.powered = True

def button_callback(*args, **kwargs):
    character_dict = args[1]
    if 'Value' in character_dict:
        btn_value = int.from_bytes(character_dict['Value'], byteorder='big')
        print('Button value ', btn_value)
        if btn_value > 1:
            print('long press')
        elif btn_value > 0:
            print('short press')
        else:
            print('released')

def try_reconnect():
    print('Try to connect')
    while not ubit.connected:
        try:
            ubit.connect()
        except:
            print('Error')
        else:
            print(ubit.connected, ubit._btn_a_state.notifying)
            if not ubit._btn_a_state.notifying:
                ubit.subscribe_button_a(button_callback)
                ubit.on_disconnect = try_reconnect

try_reconnect()

print('About to launch')

try:
    ubit.run_async()
except KeyboardInterrupt:
    print('Interrupt caught')
finally:
    print('tidying up...')
    ubit.quit_async()
    ubit.disconnect()
```

In the adapter file I changed where the `add_signal_receiver`  for `properties_changed` looks. 
In that new method, if there is a `not connected` then I call the method specified in microbit `on_disconnect` property.

This did highlight a couple of issues.
1) If I subscribed to btn_a again then I would get the button_callback called multiple times. There is not a tidy way to find out the status of the subscription.
2) I need to workout what exception is thrown when it tries and fails to connect.

As a side note: I a `try`, `except` for the event loop. This seem to be working in a tidy fashion for me. Do I need a better testcase?

Please take a look and let me have your feedback.

Thanks.

PVince81, 2018-02-15:
@ukBaz I'm using central, connecting this way:
```python
    def connect(self):
        self.central.connect();
        while not self.central.services_resolved:
            sleep(0.5)
        self.central.load_gatt()
        self.read_channel.add_characteristic_cb(self._notify_callback)
        self.read_channel.start_notify()
```

how would I go to register the on_disconnect callback from there ?

ukBaz, 2018-02-15:
Thanks @PVince81 for the feedback. I will look at pushing down what I have done to the lower level files to make this easier for you. I am not expecting to get time to look at this for a couple days so below is some untested code of what I would expect.

This is based on the assumption that you have taken the updates for adapter.py in the on_disconnect branch. 

```python
    def connect(self):
        while not self.central.connected:
            try:
                self.central.connect();
            except:
                print('Error')
            else:
                while not self.central.services_resolved:
                    sleep(0.5)
                self.central.load_gatt()
                if not self.read_channel.notifying:
                    self.read_channel.add_characteristic_cb(self._notify_callback)
                    self.read_channel.start_notify()
                    self.central.dongle.on_disconnect = self.connect # call this connect method on disconnect
```

I am sure given a little bit more time and thought a better job could be done. However give it a try and see if it does what you require.

boardend, 2018-02-15:
@ukBaz, Thanks for your proposed solution. This would work for me (besides the side effects discussed in  #180), but I'm with @PVince81 that the functionality would be useful every time a device takes a central role and should therefrom be moved to `central.py` :+1:

ukBaz, 2018-02-16:
I agree this logic/code needs to be moved to a lower level. Maybe central, maybe lower. I will look at that once we have confirmed solutions.

boardend, 2018-04-30:
@WayneKeenan yes, I'm currently using the fix from this issue on my project. Everything is working fine, so I think it would be ok to merge the changes back.

@PVince81 Let me know if I can help you with testing/polishing.

ukBaz, 2021-01-01:
This was fixed with b3cfb28c87ec3e38885cc54c05df0d869dd76bfc

closed 2021-01-01 by ukBaz