ukBaz/python-bluezero #177: Receive notification events in own main loop

PVince81, 2018-01-23:
If I were to write an application with its own main loop (ex: Qt), it would be good to have a method that can be called periodically to make python-bluezero trigger the registered notification callback.

If that's not possible, the other way would be to use a separate thread just to process notification events and queue them into an array. Then from the main code, pull data from the array. Not sure if this thread idea should live in the library or not.

ukBaz, 2018-01-23:
DBus bindings and event-loops are a constant source of discussion with my fellow collaborators on this project.

My initial reaction to this issue is that it is not something that fits inside Bluezero. My reason for saying this is because the idea of Bluezero was to enable people to have an easier time getting started with Bluetooth (Low Energy) and Python. Our vision is that as people got more advanced they would transition to using the native DBus API available from BlueZ. They could then have control of event-loops and DBus bindings.
For example, would using the Qt DBus bindings make more sense in your case:
http://pyqt.sourceforge.net/Docs/PyQt4/dbus.html

I am open to hearing more if you think there is something that would fit nicely inside of the Bluezero. For now I'm going to close this issue. Please re-open if you would like to comment more.

closed 2018-01-23 by ukBaz

PVince81, 2018-01-25:
Thanks for the info. Makes sense to have design goals to limit the scope of the project. 👍 

I'll look into using DBus directly later.