ukBaz/python-bluezero #178: How to Setup an Eddystone URL Beacon on the RasPi 3

jonathanrjpereira, 2018-02-07:
How do I set up the library & install it on my Raspberry Pi 3?
Is it compatible with Python 2.7?

ukBaz, 2018-02-07:
To install the Bluezero library, then my recommendation is just to go with getting it from PyPI
 (if you do not want to clone from GitHub):
```linux
sudo pip3 install bluezero
```
As the [readme ](https://github.com/ukBaz/python-bluezero#beacon) says you will need to put Bluetooth daemon in experimental mode for the Eddystone Beacon. This is a one time setup that can be done with:
```linux
sudo sed -i '/^ExecStart.*bluetoothd\s*$/ s/$/ --experimental/' /lib/systemd/system/bluetooth.service
```
If you have an up-to-date version of Raspbian then you should be good to go

The library is not restricting itself to be Python 2.7 compatible. Things like the Eddystone scanner are only possible in a very recent version of Python. I haven't tested the beacon code with Python 2.7.
Is there a compelling reason for needing Python 2.7?

ukBaz, 2018-02-14:
@jonathanrjpereira , there is no activity on this for a week so I am closing. If there is still an issue then please re-open

closed 2018-02-14 by ukBaz