ukBaz/python-bluezero #180: Cleanup / killing makes it slower when reconnecting

PVince81, 2018-02-13:
Follow up from https://github.com/ukBaz/python-bluezero/issues/176#issuecomment-361157992 and https://github.com/ukBaz/python-bluezero/issues/176#issuecomment-364882149.

Basically @boardend and I observed that when reconnecting to the device, the communication seems much slower.

In my case, I used the Glib main loop which I cannot interrupt due to https://bugzilla.gnome.org/show_bug.cgi?id=622084#c35 so I need to kill this remotely. The next connection is slower. And the next one even slower.

@boardend can you clarify whether your watchdog setup also kills the python process remotely or is it in a thread within the same python script ?

PVince81, 2018-02-13:
Maybe the connection is not properly cleant up due to the way we interrupt the script.

Not sure whether this is a bluezero problem or DBus itself.

ukBaz, 2018-02-13:
Thanks for splitting this out. Probably a good idea.

I am not sure I can reproduce the behaviour you are observing. Do you have a testcase I can run?

boardend, 2018-02-15:
I will try to describe my current setup and share my observations:

- @PVince81 At the moment my watchdog is realized with a thread and it will create and start a new thread for each connection/communication attempt. This means that I don't reuse the bluezero micro:bit instance and still get those side effects discussed here and in #176 .
- When I kill and restart the whole script than everything seems to be fine again. This makes me believe that I could get rid of the problem by switching from a thread to a sub-process. But I don't want to deal with inter process communication and it feels like taking a sledgehammer to crack a nut anyway.
- I've noticed the same problem with button callbacks as @ukBaz describes in #176. What makes me wonder is, that the callbacks registered/used in the first session still get called from the second session, even the thread from the first session isn't alive anymore, respectively the micro:bit instance got stopped and discarded.
- After some time in the watchdog loop (e.g. after creating and connecting certain micro:bit instances) I get the following exception `dbus.exceptions.DBusException: org.freedesktop.DBus.Error.LimitsExceeded: Connection ":1.27842" is not allowed to add more match rules (increase limits in configuration file if required)`
- When the loglevel from bluezero is set to INFO you can observe that the GATT database will increase with each connection (even when a new micro:bit instance is used). And I don't think this is just because of the registered callbacks. It seems that every characteristic gets re-added until the limit is exceeded.

IMHO bluezero has to provide a mechanism for cleaning up the GATT database or the already registered entries have to be detected and skipped on reinitialization/reconnection of a micro:bit (central) instance.

ukBaz, 2018-02-15:
Thanks for posting some more information. Very helpful.

The `org.freedesktop.DBus.Error.LimitsExceeded` error does suggest this is a DBus issue with multiple subscriptions being added for DBus signals. Are you still seeing this with the latest example that is testing to see if `notifying` is already on?
I say this because I have run my test code this evening and done about 30 re-connects and I didn't see this error.

Reading up on issues around `org.freedesktop.DBus.Error.LimitsExceeded` suggests that the Python code needs to manage this. There are a number of examples where people keep subscribing to DBus signals accidentally. I can investigate if some protection can be done in the lower level functions.

I do have concerns about the use of threads, especially multiple threads. I've done a project that maintained connection to 4 peripherals and handled the notifications without threads. My reason for raising this is I am trying to get it clear in my mind what is an issue with the Bluezero library and what is external.

BlueZ's bluetoothd does most of the heavy lifting. Bluezero is doing the DBus interprocess communication to the bluetoothd. There should not be the need to destroy Python objects and recreate them when a bluetooth disconnect and reconnect. We should be able to keep using the `central` or `microbit` objects.
As for cleaning up GATT database, bluetoothd takes care of this. Bluezero doesn't  _store_ that information. It always goes to bluetoothd to get the information.

If you can post logging output that might help me reproduce your error.

It would also be helpful if you can reproduce the error with a micro:bit and a small amount of code so I can run things here.

boardend, 2018-02-16:
With https://github.com/ukBaz/python-bluezero/issues/176#issuecomment-364902112 I've tried to ask if this approach is appropriate :-) So I've just switched my application to your testcase architecture from https://github.com/ukBaz/python-bluezero/issues/176#issuecomment-365416449 and the out of range scenario works like a charm. But I'm still not able handle the micro:bit reset/power loss scenario in a way, that the application is able to recover and reconnect to the micro:bit again.

I can still write some testcases for the problems encountering while re-creating and -connecting new `microbit.Microbit` instances. But it looks like a no through road anyway?

ukBaz, 2018-02-16:
Thanks for the feedback @boardend .  Sorry it is taking a few iteration for me to understand exactly what you have.

If I have understood you correctly, there are two scenarios you are looking to (re)connect on while your app (using Bluezero) is running:
1. Peripheral coming back into range. 
2. The peripheral is power on while your app is already running.

I have focused on number 1 and it sounds like we have something working for that scenario. 

I now need to investigate number 2.

Once we have something working, I'll then focus on usability etc before committing it into the main code.

ukBaz, 2018-02-18:
I am posting some of the investigation work I have done. There are not any conclusions at this time. There are three reasons for posting it:
1. For future reference
2. Allow others to check my working out
3. Maybe trigger some ideas in others.

If I monitor the DBus with this command:
```linux
sudo busctl monitor org.bluez
```

When I switch off a micro:bit that is connected there is the following activity on the DBus:

```linux
‣ Type=signal  Endian=l  Flags=1  Version=1  Priority=0 Cookie=195
  Sender=:1.8  Path=/org/bluez/hci0/dev_F1_55_90_65_29_DC  Interface=org.freedesktop.DBus.Properties  Member=PropertiesChanged
  UniqueName=:1.8
  MESSAGE "sa{sv}as" {
          STRING "org.bluez.Device1";
          ARRAY "{sv}" {
                  DICT_ENTRY "sv" {
                          STRING "ServicesResolved";
                          VARIANT "b" {
                                  BOOLEAN false;
                          };
                  };
                  DICT_ENTRY "sv" {
                          STRING "Connected";
                          VARIANT "b" {
                                  BOOLEAN false;
                          };
                  };
          };
          ARRAY "s" {
          };
  };

```
This is the same activity as when the micro:bit goes out of range. However as we have seen there is an issue when trying to reconnect from an out-of-range versus a power cycle.

To try to handle the power cycle case I have experimented with:

```python
def big_loop():
    print('Big loop')
    if not ubit.connected:
        try_reconnect()
    return True

ubit.ubit.dongle.mainloop.add_timer(5000, big_loop)
```

This does allow for the notifications to continue on button A after a power cycle. Hower there are multiply notifications:
```linux
Button value  1
short press
Button value  1
short press
Button value  0
released
Button value  0
released
```
Although looking on the bus there is only the correct number of signals sent:
```linux
‣ Type=signal  Endian=l  Flags=1  Version=1  Priority=0 Cookie=1307
  Sender=:1.8  Path=/org/bluez/hci0/dev_F1_55_90_65_29_DC/service0019/char001a  Interface=org.freedesktop.DBus.Properties  Member=PropertiesChanged
  UniqueName=:1.8
  MESSAGE "sa{sv}as" {
          STRING "org.bluez.GattCharacteristic1";
          ARRAY "{sv}" {
                  DICT_ENTRY "sv" {
                          STRING "Value";
                          VARIANT "ay" {
                                  ARRAY "y" {
                                          BYTE 1;
                                  };
                          };
                  };
          };
          ARRAY "s" {
          };
  };

‣ Type=signal  Endian=l  Flags=1  Version=1  Priority=0 Cookie=1308
  Sender=:1.8  Path=/org/bluez/hci0/dev_F1_55_90_65_29_DC/service0019/char001a  Interface=org.freedesktop.DBus.Properties  Member=PropertiesChanged
  UniqueName=:1.8
  MESSAGE "sa{sv}as" {
          STRING "org.bluez.GattCharacteristic1";
          ARRAY "{sv}" {
                  DICT_ENTRY "sv" {
                          STRING "Value";
                          VARIANT "ay" {
                                  ARRAY "y" {
                                          BYTE 0;
                                  };
                          };
                  };
          };
          ARRAY "s" {
          };
  };

```
This does seem like the issue is with how notifications are subscribed to. I am trying to look at where this issue is as it could be with the Python, DBus or BlueZ.

I have started to have a trawl through old posts on the BlueZ mailing list. There are a couple around what sound like similar topics. For example:
https://marc.info/?l=linux-bluetooth&m=149596078420834&w=2

Investigations continue...

ukBaz, 2018-02-18:
I have continued with the idea of the timer that every 5 seconds checks if the micro:bit (peripheral) is still connected. I have also added some logic that on the first time through it registers the callback for the disconnect event and subscribing to the characteristic. On subsequent reconnections it just turns notifications back on.

This seems to be working in my testing.

@PVince81  & @boardend please take a look and see if this works for your situations.

```python
import dbus
from bluezero import microbit

class NotReady(dbus.exceptions.DBusException):
    _dbus_error_name = 'org.bluez.Error.NotReady'

class Failed(dbus.exceptions.DBusException):
    _dbus_error_name = 'org.bluez.Error.Failed'

class InProgress(dbus.exceptions.DBusException):
    _dbus_error_name = 'org.bluez.Error.InProgress'

class AlreadyConnected(dbus.exceptions.DBusException):
    _dbus_error_name = 'org.bluez.Error.AlreadyConnected'

notify_set = False

ubit = microbit.Microbit(adapter_addr='B8:27:EB:22:57:E0',
                         device_addr='F1:55:90:65:29:DC')

print(ubit.ubit.dongle.powered)
if not ubit.ubit.dongle.powered:
    print('powering on')
    ubit.ubit.dongle.powered = True

def button_callback(*args, **kwargs):
    character_dict = args[1]
    if 'Value' in character_dict:
        btn_value = int.from_bytes(character_dict['Value'], byteorder='big')
        print('Button value ', btn_value)
        if btn_value > 1:
            print('long press')
        elif btn_value > 0:
            print('short press')
        else:
            print('released')

def disconnect_handler():
    print('Disconnect event')

def try_reconnect():
    global notify_set
    print('Try to connect', ubit.connected, notify_set)
    if not ubit.connected:
        try:
            ubit.connect()
        except NotReady:
            print('Not ready')
        except Failed:
            print('Failed to connect')
        except InProgress:
            print('Connect in progress')
        except AlreadyConnected:
            print('Already connected. Disconnecting...')
            ubit.disconnect()
        except Exception as e:
            print('Broad exception', e)
            ubit.disconnect()
        else:
            print(ubit.connected, ubit._btn_a_state.notifying)
            if not ubit._btn_a_state.notifying and not notify_set:
                print('Notifying set')
                notify_set = True
                ubit.subscribe_button_a(button_callback)
                ubit.on_disconnect = disconnect_handler
            elif not ubit._btn_a_state.notifying and notify_set:
                print('Subscribed already. Start notifications')
                ubit._btn_a_state.start_notify()

def big_loop():
    print('Big loop')
    if not ubit.connected:
        try_reconnect()
    return True

try_reconnect()
# ubit.connect()

print('About to launch')
ubit.ubit.dongle.mainloop.add_timer(5000, big_loop)

try:
    ubit.run_async()
except KeyboardInterrupt:
    print('Interrupt caught')
finally:
    print('tidying up...')
    ubit.quit_async()
    ubit.disconnect()
```

boardend, 2018-02-19:
I can confirm that your proposed solution works for both scenarios (out of range/power loss) :+1: Thank you so much!
It will take some time to adapt my application. But I will report back once the migration has been done.

ukBaz, 2018-02-19:
Thanks for the quick feedback. I'll continue to look at how to push this down into the existing files so it is easier to implement for people using Bluezero

ukBaz, 2018-10-04:
I can close this?

I'll use #176 to track adding this functionality into the library

closed 2018-10-04 by ukBaz