ukBaz/python-bluezero #181: TypeError: validate_object_path() argument 1 must be str, not None

WayneKeenan, 2018-02-16:
On a fresh Stretch install (2017-11-29-raspbian-stretch) I'm bumping into this error when running the micro:bit example:

```bash

pi@raspberrypi:~/python-bluezero/examples $ python3 microbit_poll.py 
Traceback (most recent call last):
  File "microbit_poll.py", line 5, in <module>
    device_addr='E3:AC:D2:F8:EB:B9')
  File "/usr/local/lib/python3.5/dist-packages/bluezero/microbit.py", line 68, in __init__
    device_addr=device_addr)
  File "/usr/local/lib/python3.5/dist-packages/bluezero/central.py", line 30, in __init__
    self.dongle = adapter.Adapter(adapter_addr)
  File "/usr/local/lib/python3.5/dist-packages/bluezero/adapter.py", line 83, in __init__
    self.path)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 244, in __init__
    _dbus_bindings.validate_object_path(object_path)
TypeError: validate_object_path() argument 1 must be str, not None
```

This occurs with both the pip install (v0.6) and installing from a git clone of master.
The only HCI device (HCI0) is UP.

Any ideas why dbus is throwing a fit?

Thanks
Wayne

ukBaz, 2018-02-16:
If you start `bluetoothctl` in a different terminal and you list the `devices` . Is the device 'E3:AC:D2:F8:EB:B9' in the list?

My suspicion is that you need to scan and connect to the device once (probably using `bluetoothctl`)

WayneKeenan, 2018-02-19:
Yes, scanning and connecting was the trick that made it work, um, huh?

This is a surprise to me that a manual connection (that is not strictly whitelisting/pairing/bonding or other security step) needed to have occurred prior to accessing the device. 

This seems to me to be new behaviour and is this something BlueZ has introduced?

ukBaz, 2018-02-19:
I think a scan for nearby devices has always been required before connecting/pairing. If you do a scan without connecting then BlueZ removes the device from the known devices list after a period of time. The connection is just to get the device to stick in the known list.
I have not been able to come up with a better way to scan/connect with Bluezero than Bluetoothctl does. This is why I use bluetoothctl flow in the micro:bit workshop:
https://ukbaz.github.io/howto/ubit_workshop.html

closed 2018-02-20 by WayneKeenan