ukBaz/python-bluezero #183: Using Python with Bluezero Library

619kt619, 2018-03-24:
How are you ukbaz? Thank you for your code and examples

I am extremely new to the world of programming and looking to resolve an issue that I found others were having.

TypeError: validate_object_path() argument 1 must be str, not None
- Running the latest version of Stretch 
- My micro:bit is a known device in bluetoothctl
- I am in Phython 3.5.3 IDLE

I am in your first example where we are trying to send a string to the micro:bit. Here is what is being sent back whenever I attempt to code ubit.connect():

ubit.connect()
  File "/usr/local/lib/python3.5/dist-packages/bluezero/microbit.py", line 114, in connect
    self.ubit.connect()
  File "/usr/local/lib/python3.5/dist-packages/bluezero/central.py", line 84, in connect
    self.load_gatt()
  File "/usr/local/lib/python3.5/dist-packages/bluezero/central.py", line 60, in load_gatt
    chrc.resolve_gatt()
  File "/usr/local/lib/python3.5/dist-packages/bluezero/GATT.py", line 125, in resolve_gatt
    self.chrc_uuid)
  File "/usr/local/lib/python3.5/dist-packages/bluezero/dbus_tools.py", line 287, in get_methods
    return get_dbus_iface(iface, get_dbus_obj(path_obj))
  File "/usr/local/lib/python3.5/dist-packages/bluezero/dbus_tools.py", line 86, in get_dbus_obj
    return bus.get_object(constants.BLUEZ_SERVICE_NAME, dbus_path)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 244, in __init__
    _dbus_bindings.validate_object_path(object_path)
TypeError: validate_object_path() argument 1 must be str, not None

I have been following your instructions on your workshop but have not successfully finished an example after connecting my devices. any help would be appreciated 

Here is a snap shot of the bluetoothctl so you can rule out any questions to my limited abilities. Please excuse any oversights on my part

[image]

ukBaz, 2018-03-24:
Hello Kirk,

Sorry to hear you are having some issues getting started with Bluezero. The good news is that I think the issues is that you have a hex file that doesn't have all the Bluetooth capabilities in it that is expected by Bluezero.
You can get a hex file with all the required capabilities from: http://bittysoftware.com/downloads.html#bitty_blue
You will need to use the “no pairing required” hex file.

There are some more details on getting started with Bluezero and micro:bit at:
https://ukbaz.github.io/howto/ubit_workshop.html

Let me know if that fixes the issue.

I am going to open a separate issue as Bluezero should give a better error message in this situation.

Regards,
Barry

619kt619, 2018-03-24:
Okay, I have been working with a couple of programs that use LE with the pi and microbit. I figured the hex files were the same thing and only removed the pairing required aspect of the microbit. 

I was checking out Sandeep Mistry’s program that uses noble and bleno to have the devices communicate. I think his hex file is currently flashed onto my microbit. 

Thank you for the guidance and support 

Kirk Taylor II

619kt619, 2018-03-24:
It totally worked!!! I have been frustrated the past few days trying to figure this out. Thank you so much

Kirk Taylor II

ukBaz, 2018-03-24:
Great! Glad it is working for you.

For your reference the documentation of the full micro:bit BLE profile is available at:
https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html

I have written up some more detail if you want to know about things in deeper detail than you need for using Bluezero:  
https://ukbaz.github.io/howto/ubit_ble_profile.html

closed 2018-03-24 by ukBaz