ukBaz/python-bluezero #184: Handle missing BLE services gracefully

ukBaz, 2018-03-24:
The issue seen in #183 is something that should be handled more gracefully.

After connection the code tries to resolve all of the requested services, characteristics and descriptors. If that is not found on the device then a meaning for error message should be give.

My feeling is to throw an exception **NotImplementedError** with a custom message of something like 

> The requested UUID `show UUID` is not available on the remote device

boardend, 2018-04-30:
I've written a custom ROM for the microbit which doesn't start the accelerometer or compass. (I hope that this gets me a little bit more battery life)

I don't know if this is the same issue, but it would be really nice to pass the desired services to an additional microbit constructor. Currently it is only possible to use all services listed in https://github.com/ukBaz/python-bluezero/blob/master/bluezero/microbit.py#L72

ukBaz, 2018-04-30:
I think the issues are very closely related although how you solve them is probably different.

My initial thought on this was to catch the exception and give an easier to interrupt message for users. The message would be along the lines of "The xxxx service/characteristic/UUID has not been found on the remote device. Have you got the correct hex file loaded?"
The thing I can't decide in my mind is do I do the catch this at a very low level in [DBus tools](https://github.com/ukBaz/python-bluezero/blob/master/bluezero/dbus_tools.py#L79) where it will be difficult to know the service/characteristic details.
Or at the higher level [micro:bit file](https://github.com/ukBaz/python-bluezero/blob/master/bluezero/microbit.py#L117) level where the message can be more meaningful for users.
I suspect the correct answer is all the places...

I understand for your situation you want to define that there will only a subset of micro:bit services available and to run with that.
That requires the code to be structured differently.
The simple thing is that you could change the [Microbit init](https://github.com/ukBaz/python-bluezero/blob/master/bluezero/microbit.py#L60) to take an optional parameters to define if each service is required. For example:
```Python
class Microbit:
    """
    Class to simplify interacting with a micro:bit over Bluetooth Low Energy
    """
    def __init__(self, device_addr, adapter_addr=None,
                      accelerometer=True,
                      compass=True):
        """
        Initialization of an instance of a remote micro:bit
        :param device_addr: Discovered microbit device with this address
        :param adapter_addr: Optional unless you have more than one adapter
                             on your machine
        """
        self.ubit = central.Central(adapter_addr=adapter_addr,
                                    device_addr=device_addr)

        self.user_pin_callback = None
       self.accelerometer = accelerometer
       self.compass = compass
        # Micro:bit Characteristics
       if self.accelerometer:
            self._accel_data = self.ubit.add_characteristic(ACCEL_SRV,
                                                            ACCEL_DATA)
            self._accel_period = self.ubit.add_characteristic(ACCEL_SRV,
                                                              ACCEL_PERIOD)
        if self.compass:
            self._magneto_data = self.ubit.add_characteristic(MAGNETO_SRV,
                                                              MAGNETO_DATA)
            self._magneto_period = self.ubit.add_characteristic(MAGNETO_SRV,
                                                                MAGNETO_PERIOD)
            self._magneto_bearing = self.ubit.add_characteristic(MAGNETO_SRV,
                                                                 MAGNETO_BEARING)
        self._btn_a_state = self.ubit.add_characteristic(BTN_SRV,
                                                         BTN_A_STATE)
        self._btn_b_state = self.ubit.add_characteristic(BTN_SRV,
                                                         BTN_B_STATE)
        self._io_pin_data = self.ubit.add_characteristic(IO_PIN_SRV,
                                                         IO_PIN_DATA)
        self._io_ad_config = self.ubit.add_characteristic(IO_PIN_SRV,
                                                          IO_AD_CONFIG)
        self._io_pin_config = self.ubit.add_characteristic(IO_PIN_SRV,
                                                           IO_PIN_CONFIG)
        self._io_pin_pwm = self.ubit.add_characteristic(IO_PIN_SRV,
                                                        IO_PIN_PWM)
        self._led_state = self.ubit.add_characteristic(LED_SRV,
                                                       LED_STATE)
        self._led_text = self.ubit.add_characteristic(LED_SRV,
                                                      LED_TEXT)
        self._led_scroll = self.ubit.add_characteristic(LED_SRV,
                                                        LED_SCROLL)
        self._temp_data = self.ubit.add_characteristic(TEMP_SRV,
                                                       TEMP_DATA)
        self._temp_period = self.ubit.add_characteristic(TEMP_SRV,
                                                       TEMP_PERIOD)
```

My concern with this idea being implemented in Bluezero is that the library would have to update other functions to check that the service was enabled. For example:

```python

    @property
    def accelerometer(self):
        """
        Read the values of the accelerometer on the microbit
        :return: return a list in the order of x, y & z
        """
        if not self.accelerometer:
            raise NotImplementedError("Accelerometer not implement on this micro:bit!")
        # [16, 0, 64, 0, 32, 252]
        # x=0.16, y=0.024, z=-0.992
        accel_bytes = self._accel_data.value

        return tools.bytes_to_xyz(accel_bytes)
```

Any thoughts on this?

ukBaz, 2019-03-03:
With PR #205 the first part of the suggestion above has been implemented. That is, when the micro:bit instance is created it is possible to specify which services are active.
This is enough to handle situations like issue #200 easily.
Closing this issue. If there is an issue with users not getting sensible messages if the try to use services that have not been enabled then a new issue should be opened.

closed 2019-03-03 by ukBaz