ukBaz/python-bluezero #186: Receiving the data from Micro:Bit via UART BLE service

KacperKubara, 2018-07-25:
Hey,
I'm currently trying to send the data to Raspberry Pi from Micro:Bit using the UART BLE Service: [https://lancaster-university.github.io/microbit-docs/ble/uart-service/](url). Has that feature been implemented in the Micro:Bit class, cause I can't really find it? Are there any other methods which will enable me to receive a string from Micro:Bit using this API?

ukBaz, 2018-07-25:
Hey,
This isn't something that has been done as part of the Bluezero library. 
The main reason was that the library was done to match the hex file delivered by bitty software. 

I did do an experiment with the UART service which can be found at:
https://gist.github.com/ukBaz/07cb8e67c1888bdb97c3f22272b6a263
This does take quite a different approach to how things are done in Bluezero so it will not work together. However it might get you started if you want just UART. Let me know if you have any feedback as it will be useful as I look at requirements for Bluezero.

KacperKubara, 2018-07-26:
Hey,
I think that adding the UART service would enable to make more customised projects with Micro:Bit. Eg. reading the data from various sensors connected to Micro:Bit and then sending the data obtained to Raspberry Pi. Anyway, I was thinking about how to add the UART service to this library and so far I came up with something like this(microbit.py): 
- Updating BLE characteristics:
`UART_SRV = '6E400001-B5A3-F393-E0A9-E50E24DCCA9E'`
`UART_TX = '6E400002-B5A3-F393-E0A9-E50E24DCCA9E'`
-declaring UART variable:
` self.uart_data = self.ubit.add_characteristic(UART_SRV, UART_TX)
 `
- Reading UART data:
`	def read_text(self):`
`		"""`
`		Reads the text sent from Micro:Bit`"""
`		text = ''`
`		text.join(chr(i) for i in self._uart_data)`		
`		return text`
Is this code reasonable or there are other parts of library that have to be adjusted as well? I've been mostly using C as my main programming language and I now I have to switch quickly to Python so I might be wrong.
Cheers, Kacper

ukBaz, 2018-07-26:
Hello Kacper,

There are a couple of things to unpick here. Firstly there needs to be a distinction between what can be done and what should be done within the Bluezero library.

BLE UART feels a little bit of a hack solution. The 'correct' thing in BLE is to create custom profiles for your needs not use a string service. If additional functionality was going to be added into the Bluezero library then maybe creating some custom profiles for common sensors would be the correct thing for Bluezero.

The BLE [Automation IO Profile/Service](https://www.bluetooth.com/specifications/gatt) always seemed to me to be a better general service to add. Although I am not aware of anyone implementing this on the micro:bit as yet. If someone did that then I would be interested in implementing it in Bluezero. Especially as this could (and should) be done outside the microbit class.

Bluezero microbit class does have the issue (#184)  where it doesn't do validation of what services are on the micro:bit. This means if someone has a different hex file than specific in the [tutorial](https://ukbaz.github.io/howto/ubit_workshop.html)
they will see issues. 

I understand that you might not care too much about the bigger picture of Bluezero library maintenance so looking at your code specifically, I don't think your read_text function would work directly like that. You would need to subscribe (get notifications) from the UART service. Maybe something like:

```python
    def subscribe_uart(self, user_callback):
        """
        Execute user_callback on UART data received
        :param user_callback: function to call on new data
        """
        self.uart_data.add_characteristic_cb(user_callback)
        self.uart_data.start_notify()
```
Not sure if you were thinking that the value from the UART would be stored in a stack that you would then pull values off of.

Did you try the code in the gist?  Did that work?

Cheers,
Barry

KacperKubara, 2018-07-27:
Hey Barry,
Thanks for the help, I didn't realise that the UART service has to be first subscribed. I have put the modified code here: [https://gist.github.com/KacperKubara/c9fbd7aa92d31ca7d81249b2d1a92556](url). I will test it and see what happens next week and I will let you know. I will have a customised hex file for it, but I will only add the UART Service so hopefully, there won't be any problems.
Kacper

ukBaz, 2018-10-04:
This has been quiet for a while so I am closing. Please re-open if required

closed 2018-10-04 by ukBaz

xiaolongchengithub, 2019-05-05:
this way is ok?