ukBaz/python-bluezero #189: example to connect to a peripheral device and read values

pohetsn, 2018-09-19:
Would you mind to provide an example for that while not using micro:bit nor Eddystone? I'm trying to connect to a peripheral from Raspberry Pi 3 B+.  Thank you

pohetsn, 2018-09-20:
Thanks for your prompt response! It's very helpful and I got it working for reading values from cpu_temperature.py (need to change the CPU_TMP_CHRC to 32-bit UUID).  However, could you also guide me how to write the callback to print out cpu temperature values when subscribing to the characteristics?  Originally I planed to write a class that inherits GATT.Characteristic class, but it seems that the Characteristic object cannot be added to the Central object by GATT.Central.add_characterstic, whose parameters are UUIDs.  I'm wondering what's a good way to proceed.  Thank you again.

ukBaz, 2018-09-20:
Great to hear you have made progress quickly.

The Central role already has capabilities to subscribe to a GATT characteristic. 
You need to add a callback on the characteristic, enable notifications and then start the event loop.

To build on my previous example:

```python
from time import sleep
from bluezero import central

class MyPeripheralDevice:
    def __init__(self, device_addr, adapter_addr=None):
        self.remote_device = central.Central(adapter_addr=adapter_addr,
                                             device_addr=device_addr)
        self._remote_charac = self.remote_device.add_characteristic(SERVICE_UUID,
                                                                    CHARAC_UUD)

    def connect(self):
        self.remote_device.connect()
        while not self.remote_device.services_resolved:
            sleep(0.5)
        self.remote_device.load_gatt()

    def disconnect(self):
        self.remote_device.disconnect()

    @property
    def value(self):
        return self._remote_charac.value

    def subscribe(self, user_callback):
        self._remote_charac.add_characteristic_cb(user_callback)
        self._remote_charac.start_notify()

    def run_async(self):
        self.remote_device.run()

if __name__ == '__main__':
    my_dev = MyPeripheralDevice(adapter_addr=ADAPTER_ADDR,
                                device_addr=DEVICE_ADDR)
    my_dev.connect()
    my_dev.subscribe(print)
    my_dev.run_async()
```

pohetsn, 2018-09-21:
Thanks so much for the minimum example! It's very helpful! The next question is how to quit it?  For example, if I want to stop notifying by pressing 'Q', what is the signal 'Q' emitted to dbus so that it can be used to stop notifying? Or bluezero has other ways to do it?  Appreciated!

ukBaz, 2018-09-21:
Excellent!  We are making good progress.

Sadly this might be where things start to slow down a little. I have not thought of an easy way for this library to interact with the event loop. 

I am not going to have time to experiment with this until next week so I will put some notes here as much for me when I pick this up again (or if anyone else wants to experiment with these ideas).

The functions I've used in the past are either to add a timeout or idle callback. E.g.
https://lazka.github.io/pgi-docs/#GLib-2.0/functions.html#GLib.idle_add
https://lazka.github.io/pgi-docs/#GLib-2.0/functions.html#GLib.timeout_add

In the past for detecting key presses I have used the Python evdev library.

I am away from a development environment at the moment so this is just a rough idea (it might not even run).

```python
from time import sleep
from bluezero import central
from gi.repository import GLib
from evdev import InputDevice, categorize, ecodes

class MyPeripheralDevice:
    def __init__(self, device_addr, adapter_addr=None):
        self.remote_device = central.Central(adapter_addr=adapter_addr,
                                             device_addr=device_addr)
        self._remote_charac = self.remote_device.add_characteristic(SERVICE_UUID,
                                                                    CHARAC_UUD)

    def connect(self):
        self.remote_device.connect()
        while not self.remote_device.services_resolved:
            sleep(0.5)
        self.remote_device.load_gatt()

    def disconnect(self):
        self.remote_device.disconnect()

    @property
    def value(self):
        return self._remote_charac.value

    def subscribe(self, user_callback):
        self._remote_charac.add_characteristic_cb(user_callback)
        self._remote_charac.start_notify()

    def run_async(self):
        self.remote_device.run()

    def stop_async(self):
        self.remote_device.quit()

# Find the keyboard device and add the path here
dev = InputDevice('/dev/input/event1')

def detect_q(dev_loop):
    event = dev.read_one():
    if event.code == ecodes.KEY_Q:
        dev_loop.stop_async()
    return True

if __name__ == '__main__':
    my_dev = MyPeripheralDevice(adapter_addr=ADAPTER_ADDR,
                                device_addr=DEVICE_ADDR)
    my_dev.connect()
    my_dev.subscribe(print)
    GLiib.idle_add(detect_q, mydev)
    my_dev.run_async()

```

pohetsn, 2018-09-22:
Hi @ukBaz , it works beautifully! All I need to change is to replace event1 by event0, and add self._remote_charac.stop_notify() in stop_async().  Maybe you can add this into the example folder? I think it could help some people.

ukBaz, 2018-09-22:
Really glad this is working well for you @pohetsn 
Thank you for the feedback
This has definitely been a learning experience for me and will make Bluezero better. Thank you for your help.
I will close this now and open another issue to track adding this learning to Bluezero.

closed 2018-09-22 by ukBaz

pohetsn, 2018-09-23:
I wonder whether you could describe how did you put all the pieces together, as clearly I couldn't figure out.  For example, how did you find out GLib.idle_add(detect_q, my_dev) was the right function? From its [documentation ](https://developer.gnome.org/pygobject/stable/glib-functions.html#function-glib--idle-add)I don't understand how you know to put my_dev as the 2nd parameter, and why is it a dev_loop.  To build up knowledge about how gi.repository or gobject work, would you please recommend some good resources to start with?  Thanks

pohetsn, 2018-09-26:
thank you! really appreciated!!

AlexisBRENON, 2020-06-05:
Hi.
If the GLib documentation is quiet elusive, why not rely on more pythonic interfaces?

Could it be possible to launch the event loop in a dedicated `threading.Thread`, hence looking for key press (or keyboard interruption) in the main thread ?

Something like:
```
from time import sleep
from bluezero import central
from gi.repository import GLib
from evdev import InputDevice, categorize, ecodes

class MyPeripheralDevice:
    [...]
    def run_async(self):
        self.remote_device.run()

    def stop_async(self):
        self.disconnect()
        self.remote_device.quit()

if __name__ == '__main__':
    my_dev = MyPeripheralDevice(adapter_addr=ADAPTER_ADDR,
                                device_addr=DEVICE_ADDR)
    my_dev.connect()
    my_dev.subscribe(print)

    event_loop_thread = Thread(target=my_dev.run_async)
    event_loop_thread.start()

    while True:
        try:
            event_loop_thread.join(5)
        except KeyboardInterrupt:
            print("Stopping event loop")
            my_dev.stop_async()
```

Or at least something like this (I did not test the code, and wrote it in the GitHub interface, not sure it is valid).