ukBaz/python-bluezero #190: Add Central async example

ukBaz, 2018-09-22:
The work on #189 created a nice example of a Central role device using the event loop.

Investigate if this could be made _cleaner_ . For example look to see if GLib.idle_add could be incorporated into the async_tools file and then made part of the Central functionality.

Ensure well documented example is created to make this easy for other users of Bluezero to find.

tnarik, 2021-06-10:
Would you recommend using the Central class from this library or the BLE_GATT one mentioned here: https://github.com/ukBaz/python-bluezero/issues/314 ?

I am setting up a Central to connect to the test Peripheral I created, all running on two Pis, to learn a bit more and to avoid the pitfalls of the smartphone implementations from nRF.

Currently I can connect to the Peripheral and read values. And I can start notifications (which somehow only trigger the Peripheral notify callback IF I read the characteristic before). But the whole mechanism is not working fully: I can see the Central device receiving data from the Peripheral (if I use `btmon`), but the Central instance is not triggered, and the start notify callback of the central itself doesn't seem to trigger either. I will take a look at the underlying implementation in detail. but thought it was worth checking with you.

ukBaz, 2021-06-10:
Either library should be usable for getting notifications in the central role.

The goal of the [BLE_GATT ](https://github.com/ukBaz/BLE_GATT) library was to be a very small library that worked in a Python Virtual Environment and uses the `pydbus` library rather than the non-pythonic `dbus` library.

Are you creating the device at both ends? Is the problem with the peripheral or the central end?
There have been issues reported previously about getting the peripheral to start notifying automatically but I've never got to the bottom of what the issue is.

tnarik, 2021-06-10:
it might be a mix. Let me share my horrible code (although I think this is not the best place to raise this, but I wasn't sure how to ask):
 https://gist.github.com/tnarik/4aa3c4a392613e078f2a2dd82033bfed

I run each file in a separate PI (picake or piw), so that they can find each other, connect and initiate notifications.

If I comment lines 74 to 80 (https://gist.github.com/tnarik/4aa3c4a392613e078f2a2dd82033bfed#file-central-py-L74-L80 ) the peripheral doesn't trigger the refresh value part of the `notify_callback` ( https://gist.github.com/tnarik/4aa3c4a392613e078f2a2dd82033bfed#file-peripheral-py-L190 ).

I think I'm creating the device here: https://gist.github.com/tnarik/4aa3c4a392613e078f2a2dd82033bfed#file-central-py-L93, using it to create the `Central`, and then configuring the characteristic callback correctly here: https://gist.github.com/tnarik/4aa3c4a392613e078f2a2dd82033bfed#file-central-py-L81

But I might be missing something.

ukBaz, 2022-06-17:
This issue has sat here for a while so I want to clean it up.

Reading back through there seems to be a couple things being talked about. The original point of the issue has been fixed I think.
I'm going to close this issue. @tnarik if you are still looking for help then please open a new ticket.

closed 2022-06-17 by ukBaz