ukBaz/python-bluezero #193: EddyStone Scanner not working after aioblescan broke it

eliotlandrum, 2018-12-13:
A [commit to aioblescan](https://github.com/frawau/aioblescan/commit/af01646434064638d4943f61f6caa27bd49745c2) broke the EddyStone scanner example. Getting this error when trying the example:

```
ERROR:asyncio:Exception in callback _SelectorSocketTransport._read_ready()
handle: <Handle _SelectorSocketTransport._read_ready()>
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
    self._callback(*self._args)
  File "/usr/lib/python3.5/asyncio/selector_events.py", line 730, in _read_ready
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.5/dist-packages/aioblescan/aioblescan.py", line 1272, in data_received
    self.process(packet)
  File "/usr/local/lib/python3.5/dist-packages/bluezero/observer.py", line 23, in _process_packet
    eddystone_data = aioblescan.EddyStone(ev)
AttributeError: module 'aioblescan' has no attribute 'EddyStone'
```

ukBaz, 2018-12-14:
Thank you for opening this issue. An even bigger thank you for debugging and submitting a pull request. I am going to close this issue as we have the pull request.

closed 2018-12-14 by ukBaz

eliotlandrum, 2018-12-17:
No problem.. I like to file an issue even if I have a PR coming up so that other people can find info if they're searching for errors.

ukBaz, 2018-12-18:
Thanks @eliotlandrum . Yes I like the style of opening an issue even if you are planning a PR.  (And thank you of #194 )

eliotlandrum, 2018-12-21:
@ukBaz - do you mind bumping the version and updating on pip? Thanks!!

ukBaz, 2018-12-21:
Sure, I'll try to do it over the weekend.

ukBaz, 2018-12-22:
Thanks @eliotlandrum for nudging me on this. It has been a year since the last release.
Bumped pypi to 0.0.7 and updated readthedocs to match. Any problems then please open a new issue