ukBaz/python-bluezero #196: Add an example for peripheral & central role simultaneously

dhruvkakadiya, 2019-01-31:
There is no example through which I can configure Bluetooth device as central and peripheral role simultaneously within same script because I have an application in which I want to collect data on RPi3 board from several peripheral role BLE sensors(so in this case RPi3 board needs to act as a central device) and then I want to collect all that data on the mobile application from RPi3 board(so in this case RPi3 board needs to act as a peripheral device).

ukBaz, 2019-02-01:
This topic has been discussed fairly recently on the BlueZ mailing list.
https://marc.info/?l=linux-bluetooth&m=154178114019339&w=2

BlueZ is what Bluezero and PyBluez are built on top of so I think the answer to this will depend a lot more on the version of BlueZ and your hardware.
Because of this I've linked to your question on PyBluez as it may be that we can share the knowledge between the two libraries.
https://github.com/pybluez/pybluez/issues/231

The idea of Bluezero has always been to make it easy for people to get started with Bluetooth rather than able to do everything. Your request seems to be a more advance use case and so I am not sure I would want to add this kind of example to Bluezero.
Having said that, a suggestion for solving this simply could be to add a second Bluetooth adaptor to the RPi3. This can be done easily with a Bluetooth USB dongle. If you did this then one could be your central device and the other be your peripheral.

dhruvkakadiya, 2019-02-01:
Thanks @ukBaz. Actually RPi3 Bluetooth hardware should be supporting multi role features as it's Bluetooth 4.2. For an example Nordic's BLE hardware(nRF51822/nRF52822) comes with Bluetooth 4.2 stack supports multirole but the main difference in Nordic's hardware and RPi3's Broadcomm bluetooth hardware is, Nordic just supports BLE while RPi3 supports classic bluetooth and BLE both. So theoretical I am thinking that RPi3's hardware should be able to support the same feature but I never got any example working for that.

Yes if I'll use another bluetooth dongle then sure it's easy to have one running as central and another as peripheral but I don't want to use another bluetooth dongle for this purpose.

ukBaz, 2019-02-03:
This isn't a priority for Bluezero so I'm going to close this. If you find there is a way to do both roles with BlueZ then a note here would be appreciated.

closed 2019-02-03 by ukBaz