ukBaz/python-bluezero #200: ValueError: Bad combination of inputs: found nothing

mavi0, 2019-02-20:
Hi, I am trying to follow the ubit workshop examples but I can't seem to get any of the examples working. The microbit has the [all functions except temperature alarm - no pairing required](http://www.bittysoftware.com/downloads.html#bitty_blue) hex file loaded and displays a C when `ubit.connect()` is called from the pi, however the script crashes with `ValueError: Bad combination of inputs: found nothing`. 

I think everything is on the current versions. The pi is running raspbian stretch, the version of bluetoothctl is 5.43. I have also tried installing the Required Packages from [here](https://github.com/ukBaz/python-bluezero/blob/master/docs/install_bluez.rst) to see if that would help. I'm not really sure what I'm doing wrong really, hopefully something simple though :) 

[image]
[image]
[image]

ukBaz, 2019-02-20:
My initial thought is that this might be linked to #184 .

In `bluetoothctl` can you connect to the microbit and do `info FC:ED:C0:94:BA::C3` and share the result. I would like to see what GATT services that image has so I can compare it with what the code is looking for:
https://github.com/ukBaz/python-bluezero/blob/master/bluezero/microbit.py#L28-L49

I also wonder if this would be the correct image to use from Bitty Software:
http://www.bittysoftware.com/downloads.html#microbit_blue
[image]

mavi0, 2019-02-20:
Unfortunately I still seem to get the same error with the alternate image. Here's the output for `info FC:ED:C0:94:BA::C3` with the microbit_blue image:

[image]

And here's the output with the original bitty blue image: 
[image]

Both the images seem to have the same output, but neither of them seem to match what the code is looking for..

ukBaz, 2019-02-20:
This is what services the Bluezero code is looking for:
```python
ACCEL_SRV = 'E95D0753-251D-470A-A062-FA1922DFA9A8'
MAGNETO_SRV = 'E95DF2D8-251D-470A-A062-FA1922DFA9A8'
BTN_SRV = 'E95D9882-251D-470A-A062-FA1922DFA9A8'
IO_PIN_SRV = 'E95D127B-251D-470A-A062-FA1922DFA9A8'
LED_SRV = 'E95DD91D-251D-470A-A062-FA1922DFA9A8'
TEMP_SRV = 'E95D6100-251D-470A-A062-FA1922DFA9A8'
```
Looking at the latest hex files from Bitty Software:

http://www.bittysoftware.com/downloads.html#bitty_blue 
has:
```python
ACCEL_SRV = 'E95D0753-251D-470A-A062-FA1922DFA9A8'
MAGNETO_SRV = 'E95DF2D8-251D-470A-A062-FA1922DFA9A8'
BTN_SRV = 'E95D9882-251D-470A-A062-FA1922DFA9A8'
LED_SRV = 'E95DD91D-251D-470A-A062-FA1922DFA9A8'
```
but missing:
```python
IO_PIN_SRV = 'E95D127B-251D-470A-A062-FA1922DFA9A8'
TEMP_SRV = 'E95D6100-251D-470A-A062-FA1922DFA9A8'
```

http://www.bittysoftware.com/downloads.html#microbit_blue
has:
```python
ACCEL_SRV = 'E95D0753-251D-470A-A062-FA1922DFA9A8'
LED_SRV = 'E95DD91D-251D-470A-A062-FA1922DFA9A8'
TEMP_SRV = 'E95D6100-251D-470A-A062-FA1922DFA9A8'
```
but missing:
```python
MAGNETO_SRV = 'E95DF2D8-251D-470A-A062-FA1922DFA9A8'
BTN_SRV = 'E95D9882-251D-470A-A062-FA1922DFA9A8'
IO_PIN_SRV = 'E95D127B-251D-470A-A062-FA1922DFA9A8'
```
This means that both images will fail. It appears bitty software has changed the images on their site.

The simplest way to fix this is probably to create our own micro:bit hex file. This can be done in https://makecode.microbit.org/ with the following:
[image]

You will need to update the project settings to not require pairing:
[image]

I will test this later when I am able to get my Raspberry Pi out. If you test it before then let me know

ukBaz, 2019-02-20:
OK, so I have been doing some testing of this and it is not working for me.

On further investigation the makecode is not creating the temperature service correctly so I am also see a failure. I will raise this with makecode and see what they say.

If you want to get something working temporarily then you could modify the microbit.py to remove the temperature service.

As reference I have used the nRF Connect app on my phone to confirm the temperature service is not setup correctly

[image]

mavi0, 2019-02-21:
That's fab, I didn't need the temperature service anyway so this is a perfect fix. Thank you very much!

ukBaz, 2019-02-21:
Great! Glad you have a way forward.

I will close this and open a different issue to track identifying a micro:bit image that works with Bluezero's microbit code (and workshop)

closed 2019-02-21 by ukBaz

viktor404, 2019-03-24:
Thank you for this . I had to change the defaults in the Microbit class constructor in microbit.py to
    pin_service=True,
    temperature_service=False
to get the BitCommander example working. (I'm using version 0.1.0 via pip3 install)