ukBaz/python-bluezero #201: Not installed correctly

sherifshsffa91, 2019-02-21:
I installed bluezero successfully but when trying to copy peripherals role to dbus using. 

sudo cp examples/ukBaz.bluezero.conf /etc/dbus-1/system.d/
It returns examples/ukBaz. bluezero.conf   not existed

ukBaz, 2019-02-21:
Ah! Yes, those instructions are bad if you have installed Bluezero with pip.  Thank you for reporting as I need to improve my packaging for pip installation.

As a workaround you can get the dbus conf file with the following commands
```
curl https://raw.githubusercontent.com/ukBaz/python-bluezero/master/examples/ukBaz.bluezero.conf > ukBaz.bluezero.conf

sudo cp ukBaz.bluezero.conf /etc/dbus-1/system.d/
```

sherifshsffa91, 2019-02-21:
yes you are right , it worked for me .

I need further support regarding,
I am trying to create peripheral with one service with one characteristic using this code: 
from bluezero import peripheral

MyServiceUUID= '0000ff00-0000-1000-8000-00805f9b34fb'
station_id_UUID='0000ff01-0000-1000-8000-00805f9b34fb'
try:
    myapp=peripheral.Application()
    myserv=peripheral.Service(station_id_UUID,True)
    mychar=peripheral.Characteristic(MyServiceUUID,['write','read'],myserv,'GattServer')
    myserv.add_characteristic(mychar)
    myapp.add_service(myserv)
    myapp.start()    
except:
    myapp.stop()
 

My problem is that it connects one time and it remains unable to connect again .

ukBaz, 2019-02-22:
Glad you have the DBus config. I will track your code questions and updating the documentation in separate issues

closed 2019-02-22 by ukBaz