ukBaz/python-bluezero #202: Update micro:bit hex information in Bluezero workshop

ukBaz, 2019-02-21:


ukBaz, 2019-02-21:
As identified in #200 the Bitty Software hex images have changed and are no longer compatible with Bluezero's micro:bit central role implementation.
This issue is made worse by #184 .  Look to find a better solution to the problem. The likely solution is to use MakeCode to create the hex file.
An open question is, should the workshop show how to create the hex? Or host the hex somewhere?

carlosperate, 2019-02-21:
I would recommend using https://github.com/lancaster-university/microbit-samples/tree/master/source/examples/bluetooth-services instead.
This PR has a bit more info: https://github.com/lancaster-university/microbit-samples/pull/27

How many services do you need? The memory allocator in the DAL has changed (to be safer) and that has had an impact on BLE memory allocations.

carlosperate, 2019-02-21:
Oh, and check if these hex files work for you: https://lancaster-university.github.io/microbit-docs/ble/profile/#all-services-enabled-hex-file

ukBaz, 2019-02-21:
Thanks @carlosperate very helpful . There are a few there to choose from!  :-)
They all seem to require pairing and don't have the "C" and "D" 

I'll do a bit more research and have a think

carlosperate, 2019-02-21:
Yeah, they have pairing enabled, but they should be showing the C and D:
https://github.com/lancaster-university/microbit-samples/blob/master/source/examples/bluetooth-services/main.cpp#L31-L39

ukBaz, 2019-02-21:
The hex file that looks of most interest to me is:
https://lancaster-university.github.io/microbit-docs/resources/BLE_All_Services_DAL_2-1-1-No-DFU.hex.zip
from:
https://lancaster-university.github.io/microbit-docs/ble/profile/#all-services-enabled-hex-file

It is not displaying the C and D on connect and disconnect. Apart from that it is working well for me

There is another change that is new. It is a characteristic under the magnetometer service:
`e95db358-251d-470a-a062-fa1922dfa9a8`
This doesn't appear to be documented anywhere apart from https://github.com/lancaster-university/microbit-dal/issues/412#issuecomment-448186568
The normally brilliant nRF connect app has it down as an **Unknown Characteristic** also.
Sending an `0x01` to it does indeed cause a calibration to be required and after that magnetometer starts working.

Recording DAPLink Firmware as this might be good to refer back to
```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 9900000036424e4500733016000000540000000097969901
HIC ID: 97969901
Auto Reset: 1
Automation allowed: 0
Daplink Mode: Interface
Interface Version: 0241
Git SHA: fa4132987102c51c110751d8bdb8a74aeb7b071b
Local Mods: 1
USB Interfaces: MSD, CDC, HID
Interface CRC: 0xe369fb16
```
Details of how to roll back to this version are here:
https://microbit.org/guide/firmware/

Upgraded to 0250 and still no C  on connection
```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 9900000036424e4500733016000000540000000097969901
HIC ID: 97969901
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0250
Git SHA: 682d8303e37355532402b8d93c4f240a3cec02a9
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Interface CRC: 0x3f2b7e12
Remount count: 0
URL: https://microbit.org/device/?id=9900&v=0250
```

carlosperate, 2019-02-22:
Thanks Barry, I've opened this ticket in the docs repository to document the new characteristic (good catch!): https://github.com/lancaster-university/microbit-docs/issues/76

I've also create an internal ticket to look into why the C and D are not being displayed.

ukBaz, 2019-03-04:
Fixed by https://github.com/ukBaz/ukBaz.github.io/commit/1e5997b281ee531705ce99da31e1d429014ce758

closed 2019-03-04 by ukBaz