ukBaz/python-bluezero #204: Can only connect one time to Peripheral

ukBaz, 2019-02-22:
yes you are right , it worked for me .

I need further support regarding,
I am trying to create peripheral with one service with one characteristic using this code: 
```
from bluezero import peripheral

MyServiceUUID= '0000ff00-0000-1000-8000-00805f9b34fb'
station_id_UUID='0000ff01-0000-1000-8000-00805f9b34fb'
try:
    myapp=peripheral.Application()
    myserv=peripheral.Service(station_id_UUID,True)
    mychar=peripheral.Characteristic(MyServiceUUID,['write','read'],myserv,'GattServer')
    myserv.add_characteristic(mychar)
    myapp.add_service(myserv)
    myapp.start()    
except:
    myapp.stop()
```
 
My problem is that it connects one time and it remains unable to connect again ,
it seems like the connection remains connected while I am trying to connect again and 
when I restart the application it gives me the following:

**setup ad_manager
setup service_manager
('Advertising service ', '0000ff01-0000-1000-8000-00805f9b34fb')
('Register Advert', dbus.ObjectPath('/ukBaz/bluezero/advertisement1977054928'))
('Register Application ', dbus.ObjectPath('/ukBaz/bluezero/application1990322768'))
GetManagedObjects
Advertisement registered
GATT application registered**

_Originally posted by @sherifshsffa91 in https://github.com/ukBaz/python-bluezero/issues/201#issuecomment-466211109_

ukBaz, 2019-02-22:
A slight side note to start with, you might want to be cautious about using the UUID values of 0000ff00-0000-1000-8000-00805f9b34fb and 0000ff01-0000-1000-8000-00805f9b34fb for custom service/characteristics because these values are within the range the Bluetooth Special Interest Group (Bluetooth SIG) has reserved.
The base UUID that is reserved is:
0000xxxx-0000-1000-8000-00805F9B34FB

Having said that, I tried your code and was able to connect, disconnect, and reconnect.
I was able to write and read the characteristic using the nRF Connect app on my phone.

What were you expecting to see that you didn't?

Have you read and done the items from the system setup documentation?
https://bluezero.readthedocs.io/en/latest/system_setup.html#using-bluezero-for-a-peripheral-role-or-beacon

sherifshsffa91, 2019-02-22:
Yes I expect to connect and disconnect lots of times but I didn't.
May it be due to installation using pip install while in the advertising folder is 
'ukBaz/bluezero/......' 
And in installation using pip there is no folder called with ukBaz/bluezero/....

ukBaz, 2019-02-22:
The system that I tested this on yesterday was a fresh system that I used pip to install Bluezero on so I would be reasonably confident pip is not the issue here.
`/ukBaz/bluezero/...`  is referring to the structure of DBus not of the library on disk so that is not an issue.
The wget in the documentation is referring to the Linux Bluetooth stack (BlueZ). I do not think you need to do that at this stage.

Are you running this peripheral code on a Raspberry Pi?
What are you using to connect to your peripheral?
With your current code, nothing will be printed out to the screen when there is a connection/disconnection event so what is giving you the impression that something is wrong?
In separate terminal run `bluetoothctl`. This will report the changing state of the Bluetooth on your peripheral. This should help us understand what is going on.
When you connect and disconnect you should see that reflected in `bluetoothctl`

ukBaz, 2019-02-23:
I have spent some more time on this trying various things and have been unable to reproduce what I think you are reporting.

Below is an example where I've added a callback for when the peripheral is written to.

```python
from bluezero import peripheral

my_service_UUID = '0000ff00-0000-1000-8000-00805f9b34fb'
station_id_UUID = '0000ff01-0000-1000-8000-00805f9b34fb'

def my_write_cb(value):
    val_int = int.from_bytes(value, byteorder='little', signed=False)
    print('New value:', val_int)
    
try:
    myapp = peripheral.Application()
    myserv = peripheral.Service(station_id_UUID, True)
    mychar = peripheral.Characteristic(my_service_UUID,
                                       ['write','read'], 
                                       myserv, 
                                       [0x00])
    mychar.add_write_event(my_write_cb)
    myserv.add_characteristic(mychar)
    myapp.add_service(myserv)
    print('Dongle alias:', myapp.dongle.alias)
    myapp.start()    
except:
    myapp.stop()
```

As I can't reproduce and you seem to have moved on to other libraries I shall close this. Please re-open if you have more information that might allow me to reproduce the problem.

closed 2019-02-23 by ukBaz