ukBaz/python-bluezero #207: cpu_temperature.py notify not automatically updating values on Bluetooth Centrals

zwnich, 2019-03-04:
Hi there,

First of all, I wanted to thank you for your work on this library. As somebody who has recently had to dive into Bluetooth programming for the first time for work and ramp up quickly, your library has made the learning curve considerably more manageable. 

First, some background before I get to my issue. My hardware target is an embedded Linux platform based on the NXP iMX6. I need to implement a GATT server on this hardware that will pair with mobile phones. I'm running Linux kernel 4.9, and blueZ 5.48. I know your README explicitly states that you work only with 5.43. I'm not convinced this is an issue, as features that were previously marked experimental in 5.43, like some of the GATT interfaces, are now marked stable in this version.

I have taken your cpu_temperature.py and added multiple characteristics to it -- some read, some write, and a few with read and notify. The read and write-only characteristics work just as they should, but the notify characteristics only update their value on mobile apps if the user manually initiates a read. I've been able to reproduce this issue with your cpu_temperature.py script, leaving it exactly as it is in the repo.

Using the Nordic nRF Connect app on both an iPhone and Android, the behavior is the same. The app can read and write characteristics just fine, and it can even toggle the Notify task thread -- the one you use `GObject.timeout_add()` -- on and off. But the value of the characteristic never updates on the app. 

I should also mention that I have seen identical behavior when running the python script -- `example-gatt-server` which is packaged witih bluez-5.48. I've also noticed the code is very similar to yours. It seems possible that the issue is not with your library at all, but perhaps blueZ. I haven't been able to trace it. My question for you is this: have you ever seen this kind of behavior, and do you have any idea what might be the cause? I've scoured the web and have yet to find anything. In my experience that usually means the problem is something dumb that the documentation would specify, but I've scoured both your docs and the blueZ gatt-api documentation. I would sincerely appreciate any help you can offer.

Thanks again for a great library and learning tool!

ukBaz, 2019-03-04:
Thanks for the nice comments and sorry to hear you are having issues.

This is the configuration I'm testing this on:
```
$ uname -a
Linux linaro-developer 4.9.56-linaro-lt-qcom #1 SMP PREEMPT Mon Oct 16 12:29:10 UTC 2017 aarch64 GNU/Linux
$ bluetoothctl -v
5.43
```
Rather unhelpfully, the `cpu_temperature.py` example is working for me. When I subscribe to notification on the Nordic nRF Connect app I see the temperature update. The only slight change is that I use the following function to generate the randomly changing values:
```python
def get_cpu_temperature():
    return random.randrange(3200, 5310, 10) / 100
```

Given that the BlueZ `example-gatt-server` code is not working for you either then I suspect there is a more fundamental issue on your system.

I wonder if the the issue might be with the event loop. Does the following example run for you?
```python
import datetime
from gi.repository import GObject

def simple_cb():
    print('It is now:', datetime.datetime.now())
    return True

GObject.timeout_add(5000, simple_cb)

mainloop = GObject.MainLoop()
mainloop.run()
```

I seem to remember there was some update to BlueZ's DBus API that Bluezero will need to handle. Can't remember off the top of my head. I'll update the ticket if I remember or get chance to install it on my system.

On a side note, Bluezero focuses on getting started and so does not talk about paring and security which might be of interest to you depending on your application

ukBaz, 2019-03-04:
I've upgraded my system to BlueZ 5.48
```
$ uname -a
Linux linaro-developer 4.9.56-linaro-lt-qcom #1 SMP PREEMPT Mon Oct 16 12:29:10 UTC 2017 aarch64 GNU/Linux
$ bluetoothctl -v
bluetoothctl: 5.48
```
I am now seeing the issue where the peripheral appears to be sending the information as it is appearing on the System DBus. However it is not appearing on the phone.

I'll continue to investigate.

ukBaz, 2019-03-04:
I thought I would try the latest while I was at it:
```
$ bluetoothctl -v
bluetoothctl: 5.50
```
This is also displaying the issue that notifications are not showing on the phone

ukBaz, 2019-03-04:
Switched back to 5.43 and the notifications are still not working on the phone.

I'll rebuild my system back to a working configuration. To get the new versions of BlueZ to build I needed to do:
`sudo apt-get build-dep bluez`
I'll take a note of what libraries that updates

zwnich, 2019-03-05:
Thank you so much for your quick response. Looks like I was wrong to assume the DBus API changes would have no effect. I did some research on the `AcquireNotify` API added in blueZ 5.46, and it doesn't appear to break the existing DBus API. It merely bypasses it and uses file descriptors to create a streaming interface for data transfer, avoiding DBus overhead. 

It's encouraging that you were able to reproduce the issue on your end. That's strange that you rolled back to 5.43 and still got the issue. If the example app continued to run, maybe `sudo apt-get build-dep bluez` updated bluez dependencies but didn't break the public APIs that version 5.43 depended on. 

Please let me know if you need any help. I'd be happy to offer my assistance. While notify not working is not a showstopper, it would simplify my task considerably.

ukBaz, 2019-03-05:
@zwnich as a suggestion...

As you are seeing the issue with the standard BlueZ example then I would suggest you also post something to the BlueZ mailing list:
http://www.bluez.org/development/lists/

A btmon trace is often something that helps them.

ukBaz, 2019-03-05:
I am doing the configuration of BlueZ with:
```
./configure --prefix=/usr \
            --mandir=/usr/share/man \
            --sysconfdir=/etc \
            --localstatedir=/var \
            --enable-experimental \
            --enable-maintainer-mode
```
This reports the following error with the setup:
```
configure: error: GLib >= 2.28 is required
```
If I was to do build-dep it would want to install the following:
```
$ apt-get build-dep bluez --dry-run
NOTE: This is only a simulation!
      apt-get needs root privileges for real execution.
      Keep also in mind that locking is deactivated,
      so don't depend on the relevance to the real current situation!
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  autoconf automake autopoint autotools-dev bison bsdmainutils check debhelper dh-autoreconf dh-strip-nondeterminism
  dh-systemd flex gettext gettext-base icu-devtools intltool-debian libarchive-zip-perl libbison-dev libcap-ng-dev
  libcroco3 libdbus-1-dev libdbus-glib-1-dev libfile-stripnondeterminism-perl libglib2.0-bin libglib2.0-dev
  libical-dev libical2 libicu-dev libpcre16-3 libpcre3-dev libpcre32-3 libpcrecpp0v5 libpipeline1 libreadline-dev
  libsigsegv2 libsubunit-dev libsubunit0 libtimedate-perl libtinfo-dev libtool libudev-dev m4 man-db pkg-config
  po-debconf zlib1g-dev
The following packages will be upgraded:
  dbus libdbus-1-3 libicu57 libncurses5 libncursesw5 libtinfo5 libudev1 udev
8 upgraded, 46 newly installed, 0 to remove and 139 not upgraded.
Inst udev [232-25+deb9u1] (232-25+deb9u9 Debian-Security:9/stable [arm64]) []
Inst libudev1 [232-25+deb9u1] (232-25+deb9u9 Debian-Security:9/stable [arm64])
Conf libudev1 (232-25+deb9u9 Debian-Security:9/stable [arm64])
Inst libncurses5 [6.0+20161126-1+deb9u1] (6.0+20161126-1+deb9u2 Debian:9.8/stable [arm64]) []
Inst libtinfo5 [6.0+20161126-1+deb9u1] (6.0+20161126-1+deb9u2 Debian:9.8/stable [arm64]) [libncursesw5:arm64 ]
Conf libtinfo5 (6.0+20161126-1+deb9u2 Debian:9.8/stable [arm64]) [libncursesw5:arm64 ]
Inst libncursesw5 [6.0+20161126-1+deb9u1] (6.0+20161126-1+deb9u2 Debian:9.8/stable [arm64])
Conf libncursesw5 (6.0+20161126-1+deb9u2 Debian:9.8/stable [arm64])
Inst bsdmainutils (9.0.12+nmu1 Debian:9.8/stable [arm64])
Inst libpipeline1 (1.4.1-2 Debian:9.8/stable [arm64])
Inst man-db (2.7.6.1-2 Debian:9.8/stable [arm64])
Inst libsigsegv2 (2.10-5 Debian:9.8/stable [arm64])
Inst m4 (1.4.18-1 Debian:9.8/stable [arm64])
Inst flex (2.6.1-1.3 Debian:9.8/stable [arm64])
Inst libpcrecpp0v5 (2:8.39-3 Debian:9.8/stable [arm64])
Inst libudev-dev (232-25+deb9u9 Debian-Security:9/stable [arm64])
Inst dbus [1.10.22-0+deb9u1] (1.10.26-0+deb9u1 Debian:9.8/stable [arm64]) []
Inst libdbus-1-3 [1.10.22-0+deb9u1] (1.10.26-0+deb9u1 Debian:9.8/stable [arm64])
Inst gettext-base (0.19.8.1-2 Debian:9.8/stable [arm64])
Inst autoconf (2.69-10 Debian:9.8/stable [all])
Inst autotools-dev (20161112.1 Debian:9.8/stable [all])
Inst automake (1:1.15-6 Debian:9.8/stable [all])
Inst autopoint (0.19.8.1-2 Debian:9.8/stable [all])
Inst libbison-dev (2:3.0.4.dfsg-1+b1 Debian:9.8/stable [arm64])
Inst bison (2:3.0.4.dfsg-1+b1 Debian:9.8/stable [arm64])
Inst libsubunit0 (1.1.0-3 Debian:9.8/stable [arm64])
Inst libsubunit-dev (1.1.0-3 Debian:9.8/stable [arm64])
Inst check (0.10.0-3+b3 Debian:9.8/stable [arm64])
Inst libtool (2.4.6-2 Debian:9.8/stable [all])
Inst dh-autoreconf (14 Debian:9.8/stable [all]) []
Inst libarchive-zip-perl (1.59-1+deb9u1 Debian:9.8/stable, Debian-Security:9/stable [all]) []
Inst libfile-stripnondeterminism-perl (0.034-1 Debian:9.8/stable [all]) []
Inst libtimedate-perl (2.3000-2 Debian:9.8/stable [all]) []
Inst dh-strip-nondeterminism (0.034-1 Debian:9.8/stable [all]) []
Inst libcroco3 (0.6.11-3 Debian:9.8/stable [arm64]) []
Inst gettext (0.19.8.1-2 Debian:9.8/stable [arm64]) []
[... 120 lines cut ...]
Searching around suggested the following:
```
$ sudo apt install libdbus-1-dev libudev-dev libical-dev libreadline-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  dbus icu-devtools libdbus-1-3 libical2 libicu-dev libicu57 libncurses5 libncursesw5 libtinfo-dev libtinfo5 libudev1
  udev
Suggested packages:
  default-dbus-session-bus | dbus-session-bus icu-doc readline-doc
The following NEW packages will be installed:
  icu-devtools libdbus-1-dev libical-dev libical2 libicu-dev libreadline-dev libtinfo-dev libudev-dev
The following packages will be upgraded:
  dbus libdbus-1-3 libicu57 libncurses5 libncursesw5 libtinfo5 libudev1 udev
8 upgraded, 8 newly installed, 0 to remove and 139 not upgraded.
Need to get 27.0 MB of archives.
After this operation, 99.2 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```
finally:
```
make -j 4 && sudo make install
```
This was [re]installing BlueZ 5.43. The notifications were not working after this.

ukBaz, 2019-03-06:
On the working system with notifications enabled, the temperature would appear in the right-hand terminal from the running python code. In the middle terminal monitoring the dbus, and then finally in the left-hand terminal which is running `btmon`
[image]
Running the experiment again after doing the steps in the [comment](https://github.com/ukBaz/python-bluezero/issues/207#issuecomment-469862614 ) above. 
There are still updates happening in the right-hand and middle terminals. There is not an update happening in the left-hand (`btmon`) terminal
[image]

ukBaz, 2019-03-08:
Have tried rebuilding BlueZ 5.43 on a Raspberry Pi and am seeing the same issue.

```
$ uname -a
Linux raspberrypi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
```

ukBaz, 2019-03-15:
@zwnich, I have had a go at reproducing this again. However this time it has built successfully.

I installed the latest Raspbian and then:
```
$ sudo apt update
$ sudo apt upgrade
$ sudo apt-get install rsync
$ sudo apt-get install python3-dbus
$ wget http://www.kernel.org/pub/linux/bluetooth/bluez-5.50.tar.xz
$ tar xf bluez-5.50.tar.xz
$ cd bluez-5.50/
$ sudo apt install libdbus-1-dev libudev-dev libical-dev libreadline-dev
$ sudo apt-get install glib2.0
$ ./configure --prefix=/usr             --mandir=/usr/share/man             --sysconfdir=/etc             --localstatedir=/var             --enable-experimental             --enable-maintainer-mode
$ make -j 4
$ sudo make install
$ sudo systemctl daemon-reload
$ sudo service bluetooth restart
```
I did the `apt install` libdbus first and then glib2.0.  I also noticed that uname -a gives a different value:
```
$ uname -a
Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
```
Be interested to hear an update from your side....

zwnich, 2019-03-18:
Hi! I apologize for the delay in getting back to you. I have resolved the issue. Turns out on my machine, bluetoothd requires admin privileges in order to write to the dbus interfaces that then send notifications out to connected mobile devices that enable them. If I ran my script with sudo, the problem went away. Since I don't want to have to do that, I modified `/etc/dbus-1/systemd/bluetooth.conf` by adding the `<policy user="developer">` block below. Now notify works without sudo. Thanks again for your persistence!

My bluetooth.conf:

`

<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>

  

  <policy user="root">
    <allow own="org.bluez"/>
    <allow send_destination="org.bluez"/>
    <allow send_interface="org.bluez.Agent1"/>
    <allow send_interface="org.bluez.MediaEndpoint1"/>
    <allow send_interface="org.bluez.MediaPlayer1"/>
    <allow send_interface="org.bluez.Profile1"/>
    <allow send_interface="org.bluez.GattCharacteristic1"/>
    <allow send_interface="org.bluez.GattDescriptor1"/>
    <allow send_interface="org.bluez.LEAdvertisement1"/>
    <allow send_interface="org.freedesktop.DBus.ObjectManager"/>
    <allow send_interface="org.freedesktop.DBus.Properties"/>
  </policy>

  <policy user="developer">
    <allow own="org.bluez"/>
    <allow send_destination="org.bluez"/>
    <allow send_interface="org.bluez.Agent1"/>
    <allow send_interface="org.bluez.MediaEndpoint1"/>
    <allow send_interface="org.bluez.MediaPlayer1"/>
    <allow send_interface="org.bluez.Profile1"/>
    <allow send_interface="org.bluez.GattCharacteristic1"/>
    <allow send_interface="org.bluez.GattDescriptor1"/>
    <allow send_interface="org.bluez.LEAdvertisement1"/>
    <allow send_interface="org.freedesktop.DBus.ObjectManager"/>
    <allow send_interface="org.freedesktop.DBus.Properties"/>
  </policy>

  <policy at_console="true">
    <allow send_destination="org.bluez"/>
  </policy>

  
  <policy group="lp">
    <allow send_destination="org.bluez"/>
  </policy>

  <policy context="default">
    <deny send_destination="org.bluez"/>
  </policy>

</busconfig>`

ukBaz, 2019-03-18:
Interesting. Thanks. I'll take a look.

ukBaz, 2019-03-18:
Running with `sudo` fixes it for me also. I am a little confused as to why the default install works and building from source use to work for me. Thanks for the feedback

closed 2019-03-29 by ukBaz