ukBaz/python-bluezero #211: Docstring should specify input not output

ukBaz, 2019-03-10:
https://github.com/ukBaz/python-bluezero/blob/2512460f1988e13fa4b5bc1497178872370c8230/bluezero/microbit.py#L458

ispite, 2019-04-13:
Hello, I'd like to contribute your issue, but I don't understand some moments, where you set output values or where I can find output values?
Does _io_pin_data.value store keys in an even element and values in an odd element? Or is it just a stub?

PS I clone python-bluezero by VisualStudio community, but it doesn't run
writes error: No module named 'bluezero'

ukBaz, 2019-04-13:
Hello @ispite,

Thanks for taking an interest in contributing.

I marked this as for "first-timers-only" as it requires changing one word in the docstring. My thought was that this would allow someone to practice [creating a fork and generating pull requests](https://gist.github.com/Chaser324/ce0505fbed06b947d962) without having to be distracted by what the code does.

Have you followed the [developer-install](https://bluezero.readthedocs.io/en/stable/developer_docs.html#developer-install) notes?
You will need to specify your forked repo. So use the path https://github.com/ispite/python-bluezero
This should ensure the module is found. 
Also, Bluezero will only work on Linux systems.

The function is difficult to test as it gets a list of values from a micro:bit connected via Bluetooth to you computer.
The `self._io_pin_data.value` gets a list of values from from the remote micro:bit. This is a variable length list of up to 19 Pin Number / Value pairs. The `pin_values` method turns those pairs into key/value of a dictionary.
More information about the source data format is available at:
https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html
In the section `IO PIN SERVICE - CHARACTERISTICS`, subsection `Pin Data`

Does that help?

ispite, 2019-04-13:
Do you have tasks for beginners, such as writing a function or class?

ukBaz, 2019-04-13:
Sorry, nothing at the moment.

ukBaz, 2019-04-28:
Closed with 2cf4a59a378f4e9682453be9c0f10bbdd3e11573

closed 2019-04-28 by ukBaz