ukBaz/python-bluezero #212: Device discovery

Furikuda, 2019-03-21:
Hi, and thanks for writing one of the very few sane bluez pylib :)

I'm trying to find a way to do Device detection, preferably by looking for a Service UUID exposed.
For now I've loop over all the Devices found (using dbus_tools.get_managed_objects()),  make a Central object with adapter, and device addresses, and then add_characteristics with the UUID I'm looking for, in a big try/except block.

Would you recommend anything better?

ukBaz, 2019-03-21:
Thanks for the kind words about the library.

Device detection is not something that the library has spent much time on as to do it well asynchronous (running with the event loop) is best. As you will see elsewhere in the closed issues, getting easy-to-understand asynchronous examples can be difficult.
If running asynchronously then you can monitor the PropertiesChanged signal and stop scanning when you find the device you are looking for.

There is the [SetDiscoveryFilter](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/adapter-api.txt?h=5.43#n48) that sounds like it should help.
Bluezero does not support it gracefully currently but if you want to have a play you can do something like the following to try it:

```python
from bluezero import adapter

bt = adapter.Adapter('02:00:AA:48:25:29')
if not bt.powered:
    bt.powered = True

discovery_filter1 = {'Transport': 'le',
                     'UUIDs': ['E95D93AF-251D-470A-A062-FA1922DFA9A8']}

bt.adapter_methods.SetDiscoveryFilter(discovery_filter1)

bt.nearby_discovery()
# clear filter
bt.adapter_methods.SetDiscoveryFilter({})
```
It did not seem to have the desired effect when I just tried.

Many people have found that the discovery and pairing is done very infrequently and is easily done with `bluetoothctl`. They then know the address of the device and just use that in the creation of the Central instance.
Can you say anything about your application and why you need to scan for a device?

Furikuda, 2019-03-23:
Thank you for the example code I'll play around with it in a couple days!
I got under the impression that the app I'm trying to replicate doesn't have the device address (or name) hardcoded, and instead uses `BluetoothAdapter.LeScanCallback` as described there [in the android guide](https://developer.android.com/guide/topics/connectivity/bluetooth-le#java) to search for devices exposing the UUID of interest.
That's why I was looking to re-implement the same kind of approach. Though writing down the SSID might save a lot of pain =)

ukBaz, 2019-03-26:
Thanks for the update.

It is true that if you are writing an app to share with other users then you would not want to hard code specific devices in the app.
You would also need some way of the user choosing which of the discovered devices they wanted to pair with. Part of that pairing process is often to _trust_ the device so a connection is automatic in the future.
This is often the point at which a GUI is added to make it a more user-friendly app.

If you wanted to go down the GUI route then running asynchronously will become more important for the GUI and your Bluetooth App.
This is probably not where the Bluezero library is going. At least not in the near future.

If you wanted to have an app without hardcoded device information in it, maybe you could have the device SSID passed to your script at invocation e.g. 
`python3 my_cool_app.py -dev_ssid "EB:F6:95:27:84:A0"`

If you are creating a Python central app with a GUI you might be better using a library like pydbus to communicate with BlueZ. I did a game with a GUI which might be a good example for you:
https://gist.github.com/ukBaz/267c40c4fe71b4dd6d6de82e62ac1bf3#file-big_button_gui-py

I am going to close this issue as I don't think Bluezero wants to handle device discovery. For the purposes of this library `bluetoothctl` is sufficient methodology.

closed 2019-03-26 by ukBaz

Furikuda, 2019-03-26:
Thanks for the insight, I will probably move towards taping into dbus directly, though I must admit that thing is very confusing to me =)
I'll probably steal bits & pieces from your code.

Not sure about a GUI, could be fun but I mostly wanted to write a library that can talk to my [device](https://github.com/Furikuda/Hackulele/blob/master/ABOUT.md#bluetooth-data).