ukBaz/python-bluezero #214: Comparing with gatt-python

Snevzor, 2019-03-28:
Dear @ukBaz,

How would you compare bluezero to gatt-python?
Both provide an abstraction layer between for official bluez D-Bus interface but I wonder if there are fundamental differences in the way this layer has been created and why certain components were chosen.

ukBaz, 2019-03-28:
There are a number of Python Bluetooth BLE libraries. I have been trying to track as many as I can at:
https://github.com/ukBaz/python-bluezero/wiki
The goal of Bluezero was to allow access to the four BLE roles with as straightforward code as possible. It looks like gatt-python is central role only.

BlueZ have provided a high level API through D-Bus and I think many of the libraries have switched to that API. When I started this library I hadn’t quite appreciated how much it was going to be about D-Bus. 
Most (if not all) the libraries using D-Bus are using python-dbus (including Bluezero)
The "Obsolete libraries" section at the bottom this link https://www.freedesktop.org/wiki/Software/DBusBindings/ says that python-dbus shouldn’t be used for new applications. Of the list it suggests, it appears pydbus is the most appropriate for BlueZ python projects. However, there are some gaps in it that library which mean it can only be used for the BLE Central role currently.

Some of the libraries on the wiki are using the HCI API but this is strongly discouraged by the BlueZ developers. They would prefer developers to use the [mgmt API ](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/mgmt-api.txt) if the library needs more control than D-Bus allows. Unfortunately I don't believe that Python sockets allow access to the correct sockets.

I hope that helps. If you have more questions about Bluezero then please ask.

Snevzor, 2019-03-28:
Thanks ukBaz! We're using gatt-python at the moment but suffer from some performance issues. We're sometimes seeing large delays on D-Bus calls. We're not seeing this using bluetoothctl. We only need the central role. If related to python-dbus then I should look into pydbus myself.

Snevzor, 2019-03-28:
Fyi, at first glance https://github.com/ccxtechnologies/adbus appears to be similar to txdbus but for asyncio instead of the Twisted framework.

ukBaz, 2019-03-28:
Yes, the frequency of updates on pydbus is a concern for me also.
GDbus does seem well supported but attempts to switch Bluezero to it have never succeeded as it seems quite low level and was too much effort to switch. Likewise, with adbus. 
If you do get anything working with those libraries I would be interested in seeing what you did.

You might also want to take a look at issue #130 which talks about performance issues.

Plus, in newer versions of BlueZ there are the new AcquireWrite and AcquireNotify methods to hand over from DBus to file descriptors
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/gatt-api.txt#n115

I'll close this for now but please open again if there is more to discuss

closed 2019-03-28 by ukBaz

Snevzor, 2019-03-28:
Thanks ukBaz. I will keep you posted!

I've looked at #130 before making this informative issue. Our issue is not related to throughput. We've experimented a lot with connection intervals and other parameters. Our issue is that we try to connect and we see nothing on D-Bus for anything between 3 (normal) and 30 seconds (not normal), nor in bluetoothd log and then the peripheral is connected. During this apparent lock we see PDU packets exchanged using a BLE sniffer, so on lower level the connection is alive but nothing is happening. We don't see this issue when manually connecting using bluetoothctl. Pretty weird. I have been suspecting D-Bus and co so far as bluetoothctl works fine.

ukBaz, 2019-03-28:
FYI: I did a game ([Project Big Button](https://ukbaz.github.io/howto/proj_big_btn.html)) where I was connected to four BLE device using pydbus. I didn't see any long delays with this. Code is at:
https://gist.github.com/ukBaz/267c40c4fe71b4dd6d6de82e62ac1bf3#file-big_button_gui-py